cjcliffe/CubicSDR #777: Error Unix build

9a4db, 2019-12-29:
Any advice?
cmake ok
make up to 44% ok than:
[ 44%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/GLExt.cpp.o
/home/q/CubicSDR/src/util/GLExt.cpp: In function ‘void initGLExtensions()’:
/home/q/CubicSDR/src/util/GLExt.cpp:66:37: error: ‘Display’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                     ^~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:66:37: note: suggested alternative: ‘_XDisplay’
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                     ^~~~~~~
                                     _XDisplay
/home/q/CubicSDR/src/util/GLExt.cpp:66:46: error: ‘dpy’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                              ^~~
/home/q/CubicSDR/src/util/GLExt.cpp:66:46: note: suggested alternative: ‘dup3’
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                              ^~~
                                              dup3
/home/q/CubicSDR/src/util/GLExt.cpp:66:51: error: ‘GLXDrawable’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                   ^~~~~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:66:73: error: expected primary-expression before ‘int’
 void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                                     ^~~
/home/q/CubicSDR/src/util/GLExt.cpp:66:85: error: expression list treated as compound expression in initializer [-fpermissive]
 glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                                          ^
/home/q/CubicSDR/src/util/GLExt.cpp:69:49: error: ‘XID’ was not declared in this scope
     void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                 ^~~
/home/q/CubicSDR/src/util/GLExt.cpp:69:63: error: expected primary-expression before ‘int’
     void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                               ^~~
/home/q/CubicSDR/src/util/GLExt.cpp:69:75: error: expression list treated as compound expression in initializer [-fpermissive]
    void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                                          ^
/home/q/CubicSDR/src/util/GLExt.cpp:71:31: error: expected primary-expression before ‘void’
     glxSwapIntervalEXTFunc = (void (*) (Display *dpy, GLXDrawable drawable, int interval)) dlsym(RTLD_DEFAULT,"glXSwapIntervalEXT");
                               ^~~~
/home/q/CubicSDR/src/util/GLExt.cpp:71:31: error: expected ‘)’ before ‘void’
/home/q/CubicSDR/src/util/GLExt.cpp:74:29: error: expected primary-expression before ‘void’
     DRI2SwapIntervalFunc = (void (*) (Display *dpy, XID drawable, int interval)) dlsym(RTLD_DEFAULT,"DRI2SwapInterval");
                             ^~~~
/home/q/CubicSDR/src/util/GLExt.cpp:74:29: error: expected ‘)’ before ‘void’
/home/q/CubicSDR/src/util/GLExt.cpp:83:24: error: ‘glXGetCurrentDisplay’ was not declared in this scope
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:83:24: note: suggested alternative: ‘wxGetX11Display’
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
                        wxGetX11Display
/home/q/CubicSDR/src/util/GLExt.cpp:84:21: error: expected ‘;’ before ‘drawable’
         GLXDrawable drawable = glXGetCurrentDrawable();
                     ^~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:85:37: error: ‘drawable’ was not declared in this scope
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                     ^~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:85:37: note: suggested alternative: ‘glEnable’
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                     ^~~~~~~~
                                     glEnable
/home/q/CubicSDR/src/util/GLExt.cpp:85:55: error: ‘glxSwapIntervalEXTFunc’ cannot be used as a function
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                                       ^
/home/q/CubicSDR/src/util/GLExt.cpp:88:24: error: ‘glXGetCurrentDisplay’ was not declared in this scope
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:88:24: note: suggested alternative: ‘wxGetX11Display’
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
                        wxGetX11Display
/home/q/CubicSDR/src/util/GLExt.cpp:89:21: error: expected ‘;’ before ‘drawable’
         GLXDrawable drawable = glXGetCurrentDrawable();
                     ^~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:90:35: error: ‘drawable’ was not declared in this scope
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                   ^~~~~~~~
/home/q/CubicSDR/src/util/GLExt.cpp:90:35: note: suggested alternative: ‘glEnable’
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                   ^~~~~~~~
                                   glEnable
/home/q/CubicSDR/src/util/GLExt.cpp:90:53: error: ‘DRI2SwapIntervalFunc’ cannot be used as a function
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                                     ^
CMakeFiles/CubicSDR.dir/build.make:950: recipe for target 'CMakeFiles/CubicSDR.dir/src/util/GLExt.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/util/GLExt.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
q@q-X756UX:~/CubicSDR/build$

vsonnier, 2019-12-29:
Hi, if you use the following change: 
https://github.com/cjcliffe/CubicSDR/pull/767/commits/0540d08c2dea79b668b32b1a6d58f235d65ce9d2
from https://github.com/cjcliffe/CubicSDR/pull/767, does it work better ?

9a4db, 2019-12-29:
Hello,
now OK,up and running, have only problem with:
	Loading modules... 
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
	Available factories...lime, null, remote
But this is another story :)
Thank You for prompt assistance...

vsonnier, 2019-12-29:
Thanks to you, it means this PR #767 is really necessary.

9a4db, 2019-12-29:
Yes, on Ubuntu 18.04 LTE it is...

[image]

9a4db, 2019-12-29:
Now both LimeNET and SDRPlay are working over the network, but bandwith is limited
for example RSP 1 can go 250 kHz only, on 500 kHz selection:
"
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
"
any advice on this matter?

vsonnier, 2019-12-30:
> SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...

Would mean that your host machine running CubicSDR is not able to process the flow in time, lacking CPU power. Which is strange because 250KSps is not much. Or maybe it is side effect of

> SDRThread::readStream(): 2. SoapySDR read blocking...

which rather means that there is not enough samples streaming from the PI.
Try to play with the `Stream settings`... CS16 seems correct in any case, but why on Earth `Buffer length` could be zero ???

9a4db, 2019-12-30:
Thank you, must try to play with that settings....

vsonnier, 2020-01-23:
The building issue is solved, the rest is user-specific problems, closing.

closed 2020-01-23 by vsonnier