cjcliffe/CubicSDR #778: CubicSDR build failure?

cjbrown14, 2019-12-30:
CubicSDR build error.  Help please.

CubicSDR was rock solid on a sadly departed MacBook.  I've downloaded AppImages on this new i7 Linux computer but found them very buggy and get lots of crashes.  This was my motivation for building from source.

Here's my setup:
Linux version 4.15.0-54-generic (buildd@lgw01-amd64-014) (gcc version 7.4.0 (Ubu
ntu 7.4.0-1ubuntu1~18.04.1)) #58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019

Linux Mint 19.2 Tina 64-bit

MATE 1.22.0

As a new install on new hardware, I could well be missing critical low level libraries etc??

I followed the build instructions from:

https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

All dependencies downloaded and built ok.

The issue looks to be related to X, OpenGL and MesaGL but beyond my abilities to resolve.  Probably missing some #defines somewhere?  I'm an old C hacker who finds C++ to be "write-only" :-(  The note at the end of the above Wiki on sym links to GL / Mesa does not seem to apply to my issue?

Any help on the following gratefully received.  Happy to provide more info on this system to help make this otherwise fantastic tool more robust!

Here's the compiler output:

/home/abc/CubicSDR/src/util/GLExt.cpp: In function ‘void initGLExtensions()’:
/home/abc/CubicSDR/src/util/GLExt.cpp:66:37: error: ‘Display’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                     ^~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:66:37: note: suggested alternative: ‘_XDisplay’
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                     ^~~~~~~
                                     _XDisplay
/home/abc/CubicSDR/src/util/GLExt.cpp:66:46: error: ‘dpy’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                              ^~~
/home/abc/CubicSDR/src/util/GLExt.cpp:66:46: note: suggested alternative: ‘dup3’
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                              ^~~
                                              dup3
/home/abc/CubicSDR/src/util/GLExt.cpp:66:51: error: ‘GLXDrawable’ was not declared in this scope
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                   ^~~~~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:66:73: error: expected primary-expression before ‘int’
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                                         ^~~
/home/abc/CubicSDR/src/util/GLExt.cpp:66:85: error: expression list treated as compound expression in initializer [-fpermissive]
     void (*glxSwapIntervalEXTFunc) (Display *dpy, GLXDrawable drawable, int interval) = 0;
                                                                                     ^
/home/abc/CubicSDR/src/util/GLExt.cpp:69:49: error: ‘XID’ was not declared in this scope
     void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                 ^~~
/home/abc/CubicSDR/src/util/GLExt.cpp:69:63: error: expected primary-expression before ‘int’
     void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                               ^~~
/home/abc/CubicSDR/src/util/GLExt.cpp:69:75: error: expression list treated as compound expression in initializer [-fpermissive]
     void (*DRI2SwapIntervalFunc) (Display *dpy, XID drawable, int interval) = 0;
                                                                           ^
/home/abc/CubicSDR/src/util/GLExt.cpp:71:31: error: expected primary-expression before ‘void’
     glxSwapIntervalEXTFunc = (void (*) (Display *dpy, GLXDrawable drawable, int interval)) dlsym(RTLD_DEFAULT,"glXSwapIntervalEXT");
                               ^~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:71:31: error: expected ‘)’ before ‘void’
/home/abc/CubicSDR/src/util/GLExt.cpp:74:29: error: expected primary-expression before ‘void’
     DRI2SwapIntervalFunc = (void (*) (Display *dpy, XID drawable, int interval)) dlsym(RTLD_DEFAULT,"DRI2SwapInterval");
                             ^~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:74:29: error: expected ‘)’ before ‘void’
/home/abc/CubicSDR/src/util/GLExt.cpp:83:24: error: ‘glXGetCurrentDisplay’ was not declared in this scope
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:83:24: note: suggested alternative: ‘wxGetX11Display’
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
                        wxGetX11Display
/home/abc/CubicSDR/src/util/GLExt.cpp:84:21: error: expected ‘;’ before ‘drawable’
         GLXDrawable drawable = glXGetCurrentDrawable();
                     ^~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:85:37: error: ‘drawable’ was not declared in this scope
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                     ^~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:85:37: note: suggested alternative: ‘glEnable’
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                     ^~~~~~~~
                                     glEnable
/home/abc/CubicSDR/src/util/GLExt.cpp:85:55: error: ‘glxSwapIntervalEXTFunc’ cannot be used as a function
         glxSwapIntervalEXTFunc(dpy, drawable, interval);
                                                       ^
/home/abc/CubicSDR/src/util/GLExt.cpp:88:24: error: ‘glXGetCurrentDisplay’ was not declared in this scope
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:88:24: note: suggested alternative: ‘wxGetX11Display’
         Display *dpy = glXGetCurrentDisplay();
                        ^~~~~~~~~~~~~~~~~~~~
                        wxGetX11Display
/home/abc/CubicSDR/src/util/GLExt.cpp:89:21: error: expected ‘;’ before ‘drawable’
         GLXDrawable drawable = glXGetCurrentDrawable();
                     ^~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:90:35: error: ‘drawable’ was not declared in this scope
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                   ^~~~~~~~
/home/abc/CubicSDR/src/util/GLExt.cpp:90:35: note: suggested alternative: ‘glEnable’
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                   ^~~~~~~~
                                   glEnable
/home/abc/CubicSDR/src/util/GLExt.cpp:90:53: error: ‘DRI2SwapIntervalFunc’ cannot be used as a function
         DRI2SwapIntervalFunc(dpy, drawable, interval);
                                                     ^
make[2]: *** [CMakeFiles/CubicSDR.dir/src/util/GLExt.cpp.o] Error 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [all] Error 2

vsonnier, 2019-12-30:
Perfect duplicate of https://github.com/cjcliffe/CubicSDR/issues/777, please use CubicSDR very latest `master`.

closed 2019-12-30 by vsonnier

Geremia, 2020-03-30:
@vsonnier  Can you make a new release for CubicSDR working with the wxWidgets 3.1.3?
Master works for me, but I'd like to package CubicSDR for Slackware.
thanks