cjcliffe/CubicSDR #780: AM Mode bug with rsp1a on sdrplay's pi 4 disk image

cdbrinker, 2020-01-07:
I don't have this problem running the sdrplay software on my windows machine, but running cubicsdr on their pi 4 disk image, with AM mode on any frequency I get a loud oscillation, almost like a police siren, that drowns out the transmission. Transmission is there, but obscured. This is the case for any signal strong enough that it could otherwise be heard. This doesn't happen in other modes and I can generally hear more of an AM signal in an FM mode as it doesn't cause the 'siren' to go off.

Has anyone had this problem? 

PS
I am having so much trouble with this new pi 4. The combination of the weirdness of its ARM64 architecture and the newness of raspbian buster is confounding this linux and websdr neophyte. I didn't want to use the sdrplay disk image, but haven't been able to build cubicsdr in such a way as to let any audio out through pulseaudio, though I can see the waterfall signals and log ads-b signals in dump1090.

SDRplay, 2020-01-07:
Sounds like you are in ZeroIF mode - In ZeroIF mode you need to make sure the LO and the VFO have some separation. Otherwise the DC offset correction routine will try to remove the carrier of the signal you are on if the LO and VFO are coincident.

righthalfplane, 2020-01-14:
If you are having trouble with the audio. alsoft-conf, lets you set the various audio outputs. My pi 4 had six possible outputs, I had to switch to one of the two of that worked. If you have a monitor with speakers, the only audio that will work is the speakers on the monitor. The direct audio output on the pi 4 is inactive.

install -

sudo apt-get install alsoft-conf

run -

alsoft-conf

kerniew, 2020-02-20:
I have the same same setup and same issue with my Raspi 4.  I ran the alsoft-conf application and nothing would affect the output.  I set the center frequency and the frequency offset by 200kHz and no change.  Bandwidth changes affected the rate of the audio frequency variation.  Tuning to areas with no signal eliminates the audio "police siren".  Tuning to anything with a carrier generates the "police siren". I don't have the issue tuned to a SSB signal using AM demodulation.

vsonnier, 2020-04-12:
Apparently this issue has nothing to do with the Pi or sound config, but would be related to DSP. I'm closing this up, please continue discussion in #796 about this specific problem. There are other issues about Pi installation anyway.

closed 2020-04-12 by vsonnier

SDRplay, 2020-04-14:
Anyone suffering with this using our 0.6 SD card image, here's the fix. It looks like there was a bug in liquid-dsp that has now been fixed. We've just tried it here and it's working...

Make sure all SDR applications are closed.

$ git clone https://github.com/jgaeddert/liquid-dsp
liquid-dsp$ cd liquid-dsp
liquid-dsp$ ./bootstrap.sh
liquid-dsp$ CFLAGS="-march=native -O3" ./configure --enable-fftoverride 
liquid-dsp$ make -j4
liquid-dsp$ sudo make install
liquid-dsp$ sudo ldconfig

Then restart CubicSDR

MikaelStrom, 2020-07-24:
> Anyone suffering with this using our 0.6 SD card image, here's the fix. It looks like there was a bug in liquid-dsp that has now been fixed. We've just tried it here and it's working...
>
> Make sure [...]

I just tried this with the latest version of CubicSDR (commit 8b61aa) on MacOS and the current master of liquid-dsp (commit d10e50). It does not solve the problem for me. A slight frequency offset causes whistling in the AM demodulator. It does not happen in past versions 0.2.4 and 0.2.5 (though, i did not build the older versions, i used the binaries).