cjcliffe/CubicSDR #781: Cannot add SDR devices in the ADD window.

JoNoci, 2020-01-07:
I am using a Win-10 machine, with 16Bit Red Pitaya on Ethernet.  I have tested the setup using PowerSDR, to make sure I have correct ethernet wiring, and that the IP addresses and subnet addresses are correct - all is fine, PowerSDR works with the PitatyaI then try with CubicSDR, but the opening device page does not show the Pitaya as present. When I try (manually) add, it requests the IP address, and I enter the address ( not sure exactly how - the Pitay IP is 192.168.8.105, so  I tried addr=192.168.8.105,   then just the numbers 192.168.8.105, etc).  Then the page does a refresh, but only a short grey line appears below the 'MANUAL' word on the SDR list. The Pitaya still does not show.  If I exit the software, and go look in the Config.xml file, the file shows the redpitaya and its address under the section of Manual Devices.
The IP addresses are all correct, since the setup works fine with PowerSDR.
I have tried with CubicSDR versions 1.3, 1.4, 1.5 and 1.6 - first un-install the installed version, then install the next one, all the same result.
The Pitaya in on a network hub, as well as the PC. I then disconnected the pc and pitaya from the hub, and connected the PC network cable directly to the Pitaya, setup correct IP address ( checked again that it all works with PowerSDR) and tried CubicSDR again - still nothing.
When I plug in an AIRSPY mini Dongle to the PC USB port, it comes up right away in the device list - I can start it, stop it, etc, but no Pitaya..
Thank
Joe
Can anyone help with this please?

Dantali0n, 2020-01-08:
Have you setup the RedPitaya to use the SoapyRemote driver as this is required for it to work with CubicSDR since it does not support the SCPI remote control interface.

JoNoci, 2020-01-08:
Good Day,
Thanks for replying!  I can only answer you by saying no - I am/was unaware
that I need to do that. I loaded the Pitaya with the code as instructed on
Pavel Demin's Rp Notes websites, the Alpine code
for a Hermes class SDR Transceiver. That is how it works with PowerSDR, but
as I am very new to all this, I guess there is a hole in my knowledge as to
how these things tie together!
Pavel mentions in his pages that he wrote the Pitaya code to work
with/under the updated Soapy drivers but they not mean what I thought it
does.
I will dig and search some more to see if I can find some guidelines as to
how to do this! I did try the CubicSDR user manual, but that is a little
outdated and very underpowered..

Any wise words are welcome please!

Regards
Joe

Dantali0n, 2020-01-08:
> Any wise words are welcome please!

I have done some digging, seems you might not need SoapyRemote in this case as the https://github.com/pothosware/SoapyRedPitaya driver seems to use the SCPI interface directly. I have determined this based on that SoapySDRUtil uses the ip address as parameter:

`SoapySDRUtil --probe="driver=redpitaya, addr=192.168.1.100"`

JoNoci, 2020-01-08:
I think you are correct - but unfortunately my problem persists. I try Add a device on CubicSDR. I select RedPitaya and enter the parameter addr=192.168.8.105. When the software does the refresh, the device is not displayed. Under the Manual selection there is a device added but with no text (see screen shot attached).
If I look in the config.xml file, my device and parameter are there. So it seems the error comes when the software tries to enumerate this device. 
Any ideas ?

[image]

JoNoci, 2020-01-08:
Sorry, hit the wrong button - did not mean to close the issue

Dantali0n, 2020-01-08:
Can you show the output of the following commands:
`SoapySDRUtil --probe="driver=redpitaya, addr=192.168.8.105"`
`SoapySDRUtil --info`

JoNoci, 2020-01-08:
See attached txt files.
The IP address 192.168.8.105 is definitely correct as I can access the
device using this IP with other packages.
Joe

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=redpitaya, addr=192.168.8.105

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=redpitaya
  hardware=redpitaya

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Full gain range: [0, 0] dB
  Full freq range: [0.05, 60] MHz
    RF freq range: [0.05, 60] MHz
  Sample rates: 0.02, 0.05, 0.1, 0.25, 0.5, 1.25 MSps

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Full gain range: [0, 0] dB
  Full freq range: [0.05, 60] MHz
    RF freq range: [0.05, 60] MHz
  Sample rates: 0.02, 0.05, 0.1, 0.25, 0.5, 1.25 MSps

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2020.01.04-vc14-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.1-81ca737)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.1.2-99756be)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.1-2ae84e3)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-1c1e8aa)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-3c514ce)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2019.07.0.1-cdfcb01)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (19.04.1-9d19559c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-5a9674c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/netSDRSupport.dll   (0.1.0-dc4e579)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-5a9674c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.0-e28e4f5)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.5.2-6d9bd82)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.1-5c5d950)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-14ec39e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.3.6-47972ba)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

JoNoci, 2020-01-09:
I believe I found the issue...And you were basically correct in your original surmise, I believe.
The Pitaya I use is the 16bit version, and the Pitaya code from Pavel, as used by me, is the HPSDR compatible code, which is HERMES communication protocol - This does not provide a soapy protocol, and so works  with HPSDR, PowerSDR, and not SDR#, HDSDR, CubicSDR.  Pavel's Pitaya software that works with the latter SDR's, ie, provides Soapy interface, is only available for the 14bit Pitaya version. So, the 16Bit Pitaya will not work...

I will close this issue in a few days - in case anyone wishes to comment first.

closed 2020-01-13 by JoNoci