cjcliffe/CubicSDR #782: Undefined linker references for libatomic, building for Raspberry Pi 4

af4ex, 2020-01-09:
I am building on a Raspberry Pi 4, using the CubicSDR build instructions for Linux (Debian). Got all the dependencies built and compile CubicSDR ok, but am getting undefined references for libatomic, generating many linker error messages similar to this one:
**/usr/bin/ld: SoapySDRThread.cpp:(.text+0x46d8): undefined reference to `__atomic_load_8'**

I tried adding a filepath to my libatomic.so,  creating a link entry in CMakeCache.txt.  But still got the same errors
**OTHER_LIBS:FILEPATH=/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.so**

Also find it strange that cmake created only one output directory named "x86", suggesting that it's building for Win32. Shouldn't that be "armhf", since it's building for Raspberry Pi?

Is Raspberry Pi supported?

vsonnier, 2020-01-09:
Is Raspberry Pi supported?

If no one contributes for it, no. In the meantime, the following discussion may be of intrest for you:
https://github.com/pothosware/SoapyRTLSDR/issues/41
because it looks like exactly like your problem, in the Soapy side. Once resolves, a PR will be welcome of course.

af4ex, 2020-01-09:
> ._..because it looks like exactly like your problem on the Soapy side_

Actually, I did encounter this same error when I built SoapyRTLSDR in support of CubicSDR (as per your build instructions). But I was able to resolve by inserting this OTHER_LIBS link (the same one I tried for CubicSDR which didn't work):
**-DOTHER_LIBS:FILEPATH=/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.so**
In this case, cmake added libatomic to link dependencies in CMakeCache.txt, and it worked!

But when I added this same OTHER_LIBS link to CubicSDR, cmake apparently saw this added link and rejected it with this warning:
**CMake Warning:
  Manually-specified variables were not used by the project:

    OTHER_LIBS**

So it looks like CubicSDR cmake is causing a problem by ignoring OTHER_LIBS insertions. I believe it would have resolved my issue, like it did for SoapyRTLSDR

SDRplay, 2020-01-09:
Just do this, it works for me...

cmake .. -DOTHER_LIBRARIES="-latomic"

af4ex, 2020-01-09:
... oops, did I think I hit the wrong button and closed the issue? I was just trying to go back to edit after preview. I was trying to edit the warning, which looked like this:
**CMake Warning:**
**Manually-specified variables were not used by the project:**

**OTHER_LIBS**

af4ex, 2020-01-09:
> Just do this ... -DOTHER_LIBRARIES="-latomic"

Actually, I tried that (and also -DOTHER_LIBRARIES=atomic). But I got the same warning and no fix.

SDRplay, 2020-01-09:
That's what I did to build CubicSDR for our downloadable RPi image - strange it won't work for you.

af4ex, 2020-01-09:
Yes, strange. 
I'm not a cmake expert, but I looked in CMakeLists.txt to see if I could find where the link is being rejected and warning issued. Didn't see it. Probably embedded in some subordinate script.

SDRplay, 2020-01-09:
Did you delete the build directory and redo the whole build process? That's what I did and added -DOTHER_LIBRARIES to the standard CubicSDR cmake command.

af4ex, 2020-01-09:
Yes, I rebuilt many times, to verify what was going on. 
BTW I found another gitub project which reported the same behavior about manually specified varibles being ignored:
https://github.com/conan-io/conan/issues/5816
Looks like built-in behavior from the cmake generator because the variables were not used CMakeLists.txt:
"The problem is that there is nothing in the CMakeLists.txt that uses them, and thus the warning."

I've always found it difficult to understand CMake build files because it's so high-level.  Just adding a new link reference seems to be a nightmare. I would much prefer to work with the low-level makefiles where the recipes, actors and entities are clearly visible. :-|

af4ex, 2020-01-09:
Looks like I did accidently close this issue. Can a moderator reopen it? Or do I have to resubmit?

af4ex, 2020-01-09:
Ok. I just saw the "Reopen" button! Duh.

af4ex, 2020-01-09:
So I guess the problem is caused by CubicSDR CMake scripts (invoked by CMakeList.txt) which don't check to see if the OTHER_LIBS variable is used. So it seems to be ignored. That's what it looks like to me.
Where are all the link libraries declared anyway? I can just add it there myself.

af4ex, 2020-01-09:
**SOLVED:** the problem was that in CubicSDR the label for declaring 'other libs' is OTHER_LIBRARIES, not OTHER_LIBS (as is used in SoapyRTLSDR). So it will build without errors using this cmake command line:

`cmake ../ -DOTHER_LIBRARIES:FILEPATH=/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.so -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config` 

So after building I was able to launch the user interface, with my rtlsdr2832 V3 connected and hit START. No RF ' livedata' appeared in the waterfall, but the interface works. No error messages and I can change freq etc. So I just need to read up on how to get started with CubicSDR. It looks like a very nice receiver. 

**Now two more comments, before this issue is closed.** 

1) Someone needs to decide which is correct "OTHER_LIBS" or "OTHER_LIBRARIES" so that the build settings are consistent at the community (i.e. SDR user) level. I don't even know what labels are used beyond CubicSDR and SoapyRTLSDR.

2) I remarked above that the output directory is named "x86", which implies a 32-bit Intel processor. But Raspberry Pi is a 32-bit ARM ("armhf") platform. [Yes I know Rpi 4 can do 64 bits, but the RPi community still prefers to use only 32-bit to maintain compatibility with the early armv6 devices. That will probably change with RPi 5 ]

So here is where the 'x86' label come from in CMakeLists.txt. 
```
if( CMAKE_SIZEOF_VOID_P EQUAL 8 )
    MESSAGE( "64 bit compiler detected" )
    SET( EX_PLATFORM 64 )
    SET( EX_PLATFORM_NAME "x64" )
else( CMAKE_SIZEOF_VOID_P EQUAL 8 ) 
    MESSAGE( "32 bit compiler detected" )
    SET( EX_PLATFORM 32 )
    SET( EX_PLATFORM_NAME "x86" )
endif( CMAKE_SIZEOF_VOID_P EQUAL 8 )
```
This code implies only "Intel" platforms and not even any explicit reference to 'Linux'. [But it obviously works, somehow. I built and am running it under Linux on an ARM processor]

So I think this needs to be updated so that Linux and maybe even Raspberry Pi platforms can be explicitly declared. But this is way above my pay grade ('Simple User'). 

But at least I think I have solved the 'libatomic error' build issue. :-]

af4ex, 2020-01-09:
Looking back at the previous replies I now realize that SDRPLAY was already using "OTHER_LIBRARIES", so I guess I may have manufactured this issue by using "OTHER_LIBS", which I googled searching for a fix for the libatomic error. I found this in a least two places, so I just assumed "OTHER_LIBS" was a standard label.

Sorry about that.

But I won't close issue out, until my comment on the "x86" is reviewed by a maintainer. I think it is still a valid (but separate) issue, which addresses my question above: 'Does CubicSDR support Raspberry Pi?'

af4ex, 2020-01-09:
... and I still don't understand why adding "OTHER_LIBS" as library filepath somehow solved the libatomic error for me building SoapyRTLSDR. So I think both issues above still need to be looked at.

Tnx & 73
af4ex

af4ex, 2020-01-09:
[I did most of the above posts at 3AM this morning, so my brain was a bit foggy] Obviously, SoapyRTLSDR worked because it uses "OTHER_LIBS", not "OTHER_LIBRARIES". 

The question is: which of these labels is the default "standard", if any?

IMHO, CMake is already a tad verbose. So I greatly prefer "OTHER_LIBS". But that would probably break some existing code. 

 I did two Google queries 'CMake "OTHER_LIBS"' (187 hits) and 'CMake "OTHER_LIBRARIES"' (183 hits). So they're about the same (adding to the uncertainty and ambiguity).

Perhaps the CMake experts reading this will have the best answer to this dilemma. :-]

gregheb, 2020-01-12:
i had the same linker error messages yesterday while building CubicSDR for a Pi 3b.   I found a solution here:   https://github.com/alexa/avs-device-sdk/issues/1404     

the issue was solved by adding the line SET(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic") to the end of the CMakeLists.txt file in the CubicSDR folder.   CubicSDR is now up and working on a Pi 3b but it does use a lot of CPU.

W9KJO, 2020-04-04:
I followed the 782 solution, I think. I not very good at this Linux stuff.

Now I get this

pi@raspberrypi4b1:/Downloads/CubicSDR/build $ cmake ../ -DOTHER_LIBRARIES:FILEPATH=/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.so -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=/Develop/wxWidgets-staticlib/bin/wx-config
-- Found hamlib library at: /usr/local/lib/libhamlib.so
-- Found hamlib static library at: /usr/local/lib/libhamlib.a
-- Found hamlib static modules:
-- Found hamlib include directory at: /usr/local/include
32 bit compiler detected
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
available. Run "cmake --help-policy CMP0072" for policy details. Use the
cmake_policy command to set the policy and suppress this warning.

FindOpenGL found both a legacy GL library:

OPENGL_gl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGL.so

and GLVND libraries for OpenGL and GLX:

OPENGL_opengl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libOpenGL.so
OPENGL_glx_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGLX.so

OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
CMakeLists.txt:170 (find_package)
This warning is for project developers. Use -Wno-dev to suppress it.

-- Copying /home/pi/Downloads/CubicSDR/font/.fnt to directory /home/pi/Downloads/CubicSDR/build/x86/fonts
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono16.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono27.fnt
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono36.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono64.fnt
-- Configuring file vera_sans_mono72.fnt
-- Configuring file vera_sans_mono96.fnt
-- Copying /home/pi/Downloads/CubicSDR/font/.png to directory /home/pi/Downloads/CubicSDR/build/x86/fonts
-- Configuring file vera_sans_mono12_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono27_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono36_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono64_0.png
-- Configuring file vera_sans_mono72_0.png
-- Configuring file vera_sans_mono96_0.png
-- Copying /home/pi/Downloads/CubicSDR/icon/CubicSDR.ico to directory /home/pi/Downloads/CubicSDR/build/x86
-- Configuring file CubicSDR.ico
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Downloads/CubicSDR/build
pi@raspberrypi4b1:/Downloads/CubicSDR/build $ ls
CMakeCache.txt cmake_install.cmake CubicSDR.desktop Makefile
CMakeFiles cmake_uninstall.cmake deb_post.sh x86
pi@raspberrypi4b1:/Downloads/CubicSDR/build $ cd x86
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ ls
CubicSDR.ico fonts
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ ./CubicSDR
bash: ./CubicSDR: No such file or directory
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ cd ..
pi@raspberrypi4b1:/Downloads/CubicSDR/build $ build$ sudo make install
bash: build$: command not found

What do I need to do now?

Thanks W9KJO Walter

W9KJO, 2020-04-06:
> I followed the 782 solution, I think. I not very good at this Linux stuff.
>
> Now I get this
>
> pi@raspberrypi4b1:/Downloads/CubicSDR/build $ cmake ../ [...]

Hello Again

I posted this on this thread as requested by the moderator and now I am getting not responses or help.

I need some help resolving the problem.  I followed the directions given by the Cubic SDR webpage but there is something wrong as can be seen at the bottom of the list.  It will not do the ./CubicSDR, what ever that command is supposed to do.

Any help will be greatly appreciated.

Walter

W9KJO, 2020-04-08:
I am following the Linux Build directions and get to the Build CubicSDR section,

I type in,

pi@raspberrypi4b1:/Downloads/CubicSDR/build $ cd x86
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ ls
CubicSDR.ico fonts
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ ./CubicSDR
bash: ./CubicSDR: No such file or directory
pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ cd ..
pi@raspberrypi4b1:/Downloads/CubicSDR/build $ build$ sudo make install
bash: build$: command not found

Why would the commands not be found?  I am trying to build this on a Pi4B.

What do I need to do now?

Thanks W9KJO Walter

Dantali0n, 2020-04-10:
> I am following the Linux Build directions and get to the Build CubicSDR section,
>
> I type in,
>
> pi@raspberrypi4b1:/Downloads/CubicSDR/build $ cd x86
> pi@raspberrypi4b1:/Downloads/CubicSDR/build/x86 $ [...]

You are copying the path example as part of the command `pi@raspberrypi4b1:/Downloads/CubicSDR/build $ build$ sudo make install` here you should not type `build$ sudo make install` but just `sudo make install`.

Also Github is relatively slow paced and I would suggest not sending the same reply multiple times or on multiple different topics.

W9KJO, 2020-04-10:
> You are copying the path example as part of the command `pi@raspberrypi4b1:/Downloads/CubicSDR/build $ build$ sudo make install` here you should not type `build$ sudo make install` but just `sudo [...]

Oh wow.  How did not not realize that.  LOL  Thanks.  

Well its open now.  but not signals now I'll work on that.

73