cjcliffe/CubicSDR #784: A Power Meter for CubicSDR 0.2.5

righthalfplane, 2020-01-13:
I crowbarred a power meter into CubicSDR v0.2.0-beta-rc2, because I needed it to test some antennas. I mentioned that in a post sometime ago and a person asked, if I would post the code. Here it is - Today, I ported it to CubicSDR 0.2.5.  It took  some fiddling, but I got it going. It still works well. I tried a CB at 4 watts and at 2 watts and the meter said that they were 3 db different as it should.  You can see a test at https://youtu.be/07vbIeXHukI. I have attach a patch file for the changes to CubicSDR 0.2.5. Create a folder "CubicSDR 0.2.5" and patch it with "patch -s -p0 < cubicsdr.patch". It is calibrated for the NetSDR. You just need to change the offset factor for it to work different devices. Perhaps, someone would like to put in a way to input and save offset factors for other devices and clean up the mess. I borrowed the smeter code from cutesdr.

[cubicsdr.patch.zip](https://github.com/cjcliffe/CubicSDR/files/4051936/cubicsdr.patch.zip)

cjcliffe, 2020-06-17:
@righthalfplane thanks for the updated patch; it does seem like this would be a helpful reading to display; I'll see about converting this into a PR before the codebase around it changes much more.

bakwc, 2021-09-30:
Hi! Any progress here? Power meter is very useful feature.