cjcliffe/CubicSDR #785: CubicSDR 0.2.5 performs much poor than CubicSDR 0.2.3 with the LimeSDR Mini

righthalfplane, 2020-01-16:
On Ubuntu 18.04, CubicSDR 0.2.5 could only do 4 MSPS with a LimeSDR Mini where CubicSDR 0.2.3 was able to run the  LimeSDR Mini at its maximum 40 MSPS. I made a video showing the problem -

https://youtu.be/85RKiUUFYm0

righthalfplane, 2020-01-18:
The problem effects CubicSDR 0.2.4  also.

Dantali0n, 2020-01-19:
I remember we changed some things around starting and stopping device streams when changing sample rate which were specific to Lime devices, could be related.

vsonnier, 2020-01-19:
> I remember we changed some things around starting and stopping device streams when changing sample rate which were specific to Lime devices, could be related.

Indeed, please have a look at this: #718

Hi @righthalfplane, I hope by "0.25" you indeed means the current `master` because there is no way I'm dedicating time for such old version, especially here because #718 brought enhancements after 0.25 release on the Lime side.

righthalfplane, 2020-01-19:
I was using the pre-build and system supplied versions of Cubic0.2.5. I build CubicSDR 0.2.6a from the git source (I also had to build a special version of wxWidgets to use with it - that is why I did not immediately use the git source) and it works at the 40 MSPS. It, however, does seem to have any way to set the bandwidth - it seems to be set to a bandwidth to 5 MHZ, which only shows the center of the 40 MHZ range.

vsonnier, 2020-01-21:
>  and it works at the 40 MSPS. It, however, does seem to have any way to set the bandwidth - it seems to be set to a bandwidth to 5 MHZ, 

Which means that the device is not obeying the 40Msps `Sample rate` request in Cubic...  If you compiled a `master` version of SoapySDR and everything else is up-to-date, ...etc. then an issue should be opened there: https://github.com/myriadrf/LimeSuite because Cubic is definitly capable of more than 5Msps.

righthalfplane, 2020-01-23:
Most of the SDRs have internal filters that you can set the bandwidth for independent of the sample rate. With my limeSDR mini using SDR# or SdrGlut, I can run a 40 MSPS and set the bandwidth of the internal filters to 5 MHZ, 10 MHZ, ... With CubicSDR the internal filters are stuck at 5MHZ with no way to change them. The routine "SoapySDRDevice_setBandwidth" permits you to do this - I can find no call to SoapySDRDevice_setBandwidth in CubicSDR.

vsonnier, 2020-01-23:
Indeed, CubicSDR only use`setSampleRate` to be the the only setting that controls the number of I/Q samples/s, as it should be. The `setBandwidth` is an implementation detail that most SDR devices don't need (it works on RTL-SDR, RSP, AdamPLUTO...etc... an probably all) and Cubic have not implemented them. 
If the Lime Soapy module applies a band filter by default instead of giving a full requested sample rate unfiltered , this is the problem of the module or driver not Cubic. 

In any case, that deserves a question here:  https://github.com/myriadrf/LimeSuite.

righthalfplane, 2020-01-23:
On MacOS High Sierra with the LimeSDR mini, CubicSDR 0.2.6a comes up at a very small sample rate and you can tune one station at a time, but when you try to set the sample rate to some reasonable value it hangs. CubicSDR works OK with a RTL stick and SDRPlay. SdrGlut, which also uses SoapySDR, works just fine with the LimeSDR mini on High Sierra.

Dantali0n, 2020-01-24:
Naturally we want to use Soapy as a transparent layer that exposes SDR devices in such a way that the necessary function calls we make to the devices are universal. And as of such any behavior that is not consistent across devices should be reported to the maintainer of the Soapy device library. In this case  https://github.com/myriadrf/LimeSuite.

However, it has become clear to me that this transparency is less transparent than we would like it to be(currently 7 device specific issues open). I propose that we write a small wrapping interface that can perform some device specifics upon events such as; starting, stopping. changing sample rate and changing gain. Subsequently, we can track these specifics with the maintainers upstream and refer to the specific issue in our code.

This is less than ideal but it would at least prevent users from not being able to use some specific devices until the problems are resolved upstream.

vsonnier, 2023-01-13:
Time to close this way too old issue.

closed 2023-01-13 by vsonnier