cjcliffe/CubicSDR #787: CbicSDR

serj053, 2020-01-21:
Does not detect devices. Although the installation was performed exactly according to the site's instructions. To add modules, I used the method of reusing modules and libraries from an existing installer. But the most unpleasant thing is that even the application itself installed by the installer gives the same error. Help to understand and launch the app.
[image]

vsonnier, 2020-01-21:
You may have need to install additional libraries where Cubic can find them. But there is an easier way because you use Windows: you can install the [PothosSDR distribution](https://github.com/pothosware/PothosSDR/wiki/Tutorial) (Resources / Download installer) which contains a Win64 build of the latest CubicSDR source code together with all the existing SoapySDR modules to talk to any SDR device, together with the up-to-date required libraries.

vsonnier, 2020-01-31:
@serj053 Hi ! Did you try the PothosSDR distribution, and does it work for you ? Also, please don't open multiple issues on the same theme like #790, and continue here.

serj053, 2020-01-31:
Hello. Yes, of course, I will take into account your wishes. The Photos app works great, but I'm interested in figuring out how Cubic is built and how its elements work. That's right : https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows but I collected Cubic, collected already four times and always on this place "LibUSB 1.0 required to compile rtlsdr" I stumble, maybe there are some subtleties, although it may not be my care, tomorrow I will collect everything from the beginning.

serj053, 2020-02-01:
I reassembled the Cubic again and when I reached the "libusb" item, I executed both of the proposed options, but still , but still in both cases, CMake throws an error "CMake Error at CMakeLists.txt:77 (message):  LibUSB 1.0 required to compile rtl-sdr" when executing the "librtlsdr" item. Why it might be.

vsonnier, 2020-02-01:
Copy of your previous message:

[image]

> Friends, please tell me what this error message means.

You did not filled all the paths required for `librtlsdr`: https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows#librtlsdr, here visibly the path to THREADS_PTHREADS_INCLUDE_DIR was not filled. 

Just RTFM starting from here: https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows#pthreads-win32 and below, and don't skip steps you will get there eventually.

serj053, 2020-02-01:
Thank you for your help, I am so risking soon to master the English language perfectly)). But unfortunately I have no success. I tried to show how I completed the previous points
[image]

[image]
Following your comment didn't fix the situation. When I run CMake, the following lines should appear in the interface according to the description map - " LIBUSB_INCLUDE_DIRS ", "THREADS_PTHREADS_WIN32_LIBRARY" and "LIBUSB_LIBRARIES", but they do not appear. When I enter these lines through the "ADD" operation (STRING + FILEPATH), they disappear after configuration. I guess I should go to sleep and try again while no one is around.
[image]

[image]

vsonnier, 2022-01-15:
The Wiki entry https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows has a detailed step-by-step procedure that could work for everybody now.

closed 2022-01-15 by vsonnier