cjcliffe/CubicSDR #788: Where can I find a detailed description of the Cubic SDR

serj053, 2020-01-21:
Is there a map of the CubicSDR application where you can see which file is responsible for which functionality. This is an interesting application and I want to understand it in detail in order to create a small laboratory for studying digital signal processing.

vsonnier, 2020-01-21:
Hi @serj053, thank you for your interest ! Please read here : #756 (and follow the links described there) for pointers on how to use Cuubic.

closed 2020-01-21 by vsonnier

Dantali0n, 2020-01-21:
For what it is worth I do not think this question was about "using" Cubic but is about it's architecture. There is no real documentation on it's component architecture but I can tell you it is heavily multi-threaded and relatively easy to step through all the code using a modern IDE such as Clion.

vsonnier, 2020-01-21:
> Is there a map of the CubicSDR application where you can see which file is responsible for which functionality. 

Sorry @serj053 , I didn't read in detail. 
I've no tim to explain the architecture details, but you may start with  `SDRThread::readStream()` : this is where the I/Q samples are streamed. 
On one side, you can the follow all the output queues like `iqDataOutQueue->try_push(dataOut)` down to where they are really demodulated. 
On the other side, backtracking from  `SDRThread` will show you how it is set up which eventually ends in the UI side of things.

@Dantali0n is right, Cubic is basically made of components running `IOThread` threads piping data to each other through `ThreadBlockingQueue`: Follow all their paths back and forth and you will see all.

serj053, 2020-01-21:
Thanks for the tips. I will start mastering CubucSDR.