cjcliffe/CubicSDR #793: 0.2.6a and hamlib net device

fyoory, 2020-02-23:
Hi,

I am looking at using hamlibs net device here  on Mint 19.3.   I have rigctl enabled via systemd as /etc/systemd/system/kx3RigControl.service

```
 HDSDR  $   cat /etc/systemd/system/kx3RigControl.service 
[Unit]
Description=KX3 RIGCTLD
After=network.target

[Service]
Type=simple
ExecStart=/usr/bin/rigctld -r /dev/ttyUSB0 -s 38400 -m 229 
Restart=on-failure

[Install]
WantedBy=multi-user.target

```

This should create a systemd service that anything that can open a port to this can use it and be pretty minimalistic to run it as a system service ready to go at boot time, assuming the serial device is plugged in.

Rigctl seems to work ok here from wsjtx or rigctl command line.

```
 fury  nitro5  ~  $  sudo sockstat | grep rigctl
root     rigctld              4449     tcp4   *:4352                    *:*                       LISTEN
root     rigctld              4449     tcp4   127.0.0.1:4352            127.0.0.1:60308           ESTABLISHED
 fury  nitro5  ~  $  ps auxw | grep 4449
root      4449  0.0  0.0 180452  4824 ?        Ssl  14:41   0:00 /usr/bin/rigctld -r /dev/ttyUSB0 -s 38400 -m 229
 fury  nitro5  ~  $  rigctl -m 2 f #  Outputs the radios present frequency in kHZ.
14105000

```

However, I cannot make this talk to rigctld.  Whenever setting the port to 127.0.0.1:4352 CubicSDR  will report:
```
Rig thread starting.
Failed to connect to 127.0.0.1:4352
Rig failed to init. 
````
CubicSDR using 0.2.6a.   

I would love to figure out how to resolve this.  

Anywho, I am working out of github for cubic and soapyaudio so I can try whatever you might need.

Getting a version of this to run fully would complete my linux amateur radio setup.

Please let me know when you can what else you may need but for now here is my 
[CMakeCache.txt](https://github.com/cjcliffe/CubicSDR/files/4242673/CMakeCache.txt)

fyoory, 2020-02-23:
Ok I guess I have a lucky break today.  I was able to get it working.  The key was that the port dialog in cubic SDR can only contain the ip/hostname, not the PORT.  This should probably be adjusted so you can specify something custom if need be.

cjcliffe, 2020-06-17:
@fyoory thanks for the submission; it looks like there are several issues with specifying USB and possibly Net devices for hamlib -- seems like we might need a more comprehensive configuration dialog for the hamlib parameters.  I'll link this to issue #757 as it will likely be a related update.