cjcliffe/CubicSDR #795: Cpack error when building Cubicsdr

ah1102, 2020-03-05:
I do not often compile programs from source. Can someone tell me how to fix this? No .dylib libraries added to /CubicSDR.app/Contents/MacOS

```
Apple@iMac-Apple build % cpack
CPack: Create package using DragNDrop
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /opt/local/bin/install_name_tool: can't open input file: /Users/Apple/CubicSDR/build/x64/CubicSDR.app/Contents/Frameworks/libgcc_s.1.dylib for writing (Permission denied)
error: /opt/local/bin/install_name_tool: can't lseek to offset: 4096 in file: /Users/Apple/CubicSDR/build/x64/CubicSDR.app/Contents/Frameworks/libgcc_s.1.dylib for writing (Bad file descriptor)
error: /opt/local/bin/install_name_tool: can't write new headers in file: /Users/Apple/CubicSDR/build/x64/CubicSDR.app/Contents/Frameworks/libgcc_s.1.dylib (Bad file descriptor)
error: /opt/local/bin/install_name_tool: can't close written on input file: /Users/Apple/CubicSDR/build/x64/CubicSDR.app/Contents/Frameworks/libgcc_s.1.dylib (Bad file descriptor)
CMake Error at /opt/local/share/cmake-3.15/Modules/BundleUtilities.cmake:915 (message):
  Command failed:

   '/opt/local/bin/install_name_tool' '-id' '@executable_path/../Frameworks/libgcc_s.1.dylib' '/Users/Apple/CubicSDR/build/x64/CubicSDR.app/Contents/Frameworks/libgcc_s.1.dylib'
Call Stack (most recent call first):
  /opt/local/share/cmake-3.15/Modules/BundleUtilities.cmake:989 (fixup_bundle_item)
  /Users/Apple/CubicSDR/build/cmake_install.cmake:39 (fixup_bundle)

CPack Error: Error when generating package: CubicSDR

Apple@iMac-Apple build % ls -l x64/CubicSDR.app/Contents/MacOS 
total 16104
-rwxr-xr-x  1 Apple  staff  8243040  4 мар 18:37 CubicSDR
```
[image]

MountainLogic, 2020-03-05:
Building on OS X can be a bit of a challenge with three different ways (package managers) to get libs on the box and they each put things in different places. It is best to choose only one:  brew, MacPorts or just building from scratch. It is best to stick to one as much as practical. For the whole SDR on Mac world, I’ve found this a substantial challenge as no one way has a simple path to everything you might want. I’m on a bus right now, but there are several stackoverflow posts on resolving dylib path issues, IORC, mostly with sim links. Between where things live and conflicting version/dependencies having two package managers is a pain. Honestly, for SDR/RF I just moved my windows bootcamp to parallels and installed Ubuntu in the bootcamp partition and I could not be happier.  

At work I’m doing proof of concept RF/low level networking development on a Raspberry PI4 that I put Kali Linux on and except for a slightly unstable UI it is a no brainer from a load/packages/compile point of view. 

It also looks like you separately need to load the soapy drivers. I believe there are some brew/ports to handle this. Should be easy.

SDRplay, 2020-03-05:
I presume you have tried...

`sudo cpack`

works everytime for me.

Andy

ah1102, 2020-03-05:
Thanks, I’ll check it now. Can someone else say how to remove cmake 3.15? I removed all cmake packages from brew and macports, but it's still on the system. I have no idea where he came from.

SDRplay, 2020-03-05:
You possibly need to back up a bit as there look to be a few issues there. Firstly, you are building CubicSDR 0.2.6 which is indeed the latest version - but have you built the prerequisites? that cmake window is complaining about not seeing the wxWidgets (although I'll be honest I have no idea what that cmake window is doing as I only ever run command line build tools).

Also, SoapySDR 0.7 is old now which tells me you're not building that from source. SoapySDR 0.8 is what we're building.

For me, the only tested way to get CubicSDR built and running correctly is to follow the build instructions on the wiki and build everything from source.

If you don't have the dev files for SoapySDR then you won't be able to build the SoapySDR libraries such as SoapySDRPlay

So, what I do is to build everything from source in this order...

SoapySDR
SoapySDRPlay (or whatever Soapy framework libraries you need)
liquid-dsp
wxWidgets
CubicSDR

If you are trying to get CubicSDR working with a SDRplay device, then we provide downloads of both the SoapySDR framework and CubicSDR from our website to save you building from source.

Hope some of that helps,

Best regards,

Andy

ah1102, 2020-03-05:
I installed SoapySDR from source

```
Apple@iMac-Apple build % sudo make install
Password:
[ 53%] Built target SoapySDR
[ 63%] Built target SoapySDRUtil
[ 68%] Built target TestKwargsMarkup
[ 73%] Built target TestConvertTypes
[ 78%] Built target TestFormatParser
[ 82%] Built target TestTimeConversion
[ 85%] Built target docs
[ 87%] Built target SoapySDR_swig_compilation
[ 92%] Built target _SoapySDR
[ 95%] Built target SoapySDR3_swig_compilation
[100%] Built target _SoapySDR3
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Up-to-date: /usr/local/include/SoapySDR
-- Up-to-date: /usr/local/include/SoapySDR/Time.h
-- Up-to-date: /usr/local/include/SoapySDR/ConverterPrimitives.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.h
-- Up-to-date: /usr/local/include/SoapySDR/Formats.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.h
-- Up-to-date: /usr/local/include/SoapySDR/Time.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Types.h
-- Up-to-date: /usr/local/include/SoapySDR/Registry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/include/SoapySDR/Constants.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.hpp
-- Up-to-date: /usr/local/include/SoapySDR/ConverterRegistry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/include/SoapySDR/Types.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.0.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.dylib
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Up-to-date: /usr/local/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib/python2.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python2.7/site-packages/SoapySDR.py
-- Installing: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/SoapySDR.py
```

SDRplay, 2020-03-05:
You've got 0.7 installed from somewhere - it's in /opt/local which may mean that it's from brew or macports? You can see that 0.8 installs into /usr/local

If it was me I would purge 0.7 from the machine.

Best regards,

Andy

ah1102, 2020-03-05:
Thanks! You're right

```
Apple@iMac-Apple build % brew uninstall SoapySDR
Error: No such keg: /usr/local/Cellar/soapysdr
Apple@iMac-Apple build % sudo port  uninstall SoapySDR
Password:
The following versions of SoapySDR are currently installed:
 1) SoapySDR @0.7.0_0
 2) SoapySDR @0.7.1_0
 3) SoapySDR @0.7.1_1 (active)
Enter option(s) [1-3/all]: all
--->  Uninstalling SoapySDR @0.7.0_0
--->  Uninstalling SoapySDR @0.7.1_0
Note: It is not recommended to uninstall/deactivate a port that has dependents as it breaks the dependents.
The following ports will break:
 gr-osmosdr @20170704_3
 gr-osmosdr @20170704_7
Continue? [y/N]: y
Warning: Uninstall forced.  Proceeding despite dependencies.
--->  Deactivating SoapySDR @0.7.1_1
--->  Cleaning SoapySDR
--->  Uninstalling SoapySDR @0.7.1_1
```

ah1102, 2020-03-06:
There is nothing on the specified path /usr/local/lib/SoapySDR
```
Apple@iMac-Apple build % sudo make
Password:
[ 53%] Built target SoapySDR
[ 63%] Built target SoapySDRUtil
[ 68%] Built target TestKwargsMarkup
[ 73%] Built target TestConvertTypes
[ 78%] Built target TestFormatParser
[ 82%] Built target TestTimeConversion
[ 85%] Built target docs
[ 87%] Built target SoapySDR_swig_compilation
[ 92%] Built target _SoapySDR
[ 95%] Built target SoapySDR3_swig_compilation
[100%] Built target _SoapySDR3
Apple@iMac-Apple build % sudo make install
[ 53%] Built target SoapySDR
[ 63%] Built target SoapySDRUtil
[ 68%] Built target TestKwargsMarkup
[ 73%] Built target TestConvertTypes
[ 78%] Built target TestFormatParser
[ 82%] Built target TestTimeConversion
[ 85%] Built target docs
[ 87%] Built target SoapySDR_swig_compilation
[ 92%] Built target _SoapySDR
[ 95%] Built target SoapySDR3_swig_compilation
[100%] Built target _SoapySDR3
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Up-to-date: /usr/local/include/SoapySDR
-- Up-to-date: /usr/local/include/SoapySDR/Time.h
-- Up-to-date: /usr/local/include/SoapySDR/ConverterPrimitives.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.h
-- Up-to-date: /usr/local/include/SoapySDR/Formats.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.h
-- Up-to-date: /usr/local/include/SoapySDR/Time.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Types.h
-- Up-to-date: /usr/local/include/SoapySDR/Registry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/include/SoapySDR/Constants.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.hpp
-- Up-to-date: /usr/local/include/SoapySDR/ConverterRegistry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/include/SoapySDR/Types.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.0.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.dylib
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Up-to-date: /usr/local/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib/python2.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python2.7/site-packages/SoapySDR.py
-- Installing: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/SoapySDR.py
```

ah1102, 2020-03-06:
Now there are some problems with installing soapyRTLSDR when executing the command from the instruction

```
Apple@iMac-Apple Dev % sudo port install rtl-sdr   
--->  Fetching archive for libtool
--->  Attempting to fetch libtool-2.4.6_6+universal.darwin_19.x86_64.tbz2 from http://nue.de.packages.macports.org/libtool
--->  Attempting to fetch libtool-2.4.6_6+universal.darwin_19.x86_64.tbz2 from http://lil.fr.packages.macports.org/libtool
--->  Attempting to fetch libtool-2.4.6_6+universal.darwin_19.x86_64.tbz2 from https://packages.macports.org/libtool
Error: libtool cannot be installed for the configured universal_archs 'x86_64 i386' because the arch(s) 'i386' are not supported.
Error: Unable to execute port: upgrade libusb failed
```

ah1102, 2020-03-06:
It turns out that soapyRTLSDR should be on the computer? This is very strange, since cubicsdr 0.2.5 starts up and sees devices without any errors

cjcliffe, 2020-06-17:
@ah1102 thanks for the suggestions; I see you have an issue in for bookmarks on the spectrum, feel free to add one for the network audio request (is there a standard protocol for that?)

If I understand what manual 'gain on the signal spectrum' is I think you can accomplish that by right-dragging on the spectrum to perform a manual scale adjustment.

closed 2020-06-17 by cjcliffe

ah1102, 2020-06-18:
> - add audio output via ip to the network, for example, 127.121.1.1:7355

GQRX uses the UDP protocol, and the DSD decoder is successfully connected to the program, in the settings of which it was specified to listen to the network address

[image]