cjcliffe/CubicSDR #797: Pls add a bookmark display on the spectrum.

ah1102, 2020-03-11:
Please add a constant display of bookmarks on the spectrum, as is done in gqrx.

[image]
[image]

ah1102, 2020-04-13:
cjclife, please review the bookmark system and display them on the frequency spectrum. This is not hard to do! And how convenient it would be!

cjcliffe, 2020-06-17:
@ah1102 thanks, seems like a good idea.  Will keep this on the list as an enhancement

ah1102, 2020-06-18:
I am very glad that you did not abandon your wonderful project! Your program is the most convenient way to work with sdr on linux and macos.

When working with memory channels, I must use the gqrx program, since this program made their visualization convenient. I would like to see such conveniences in cubicsdr, since it is a very convenient and beautiful program, except for bookmarks.

tggo, 2020-07-27:
if the bookmarks can be controlled via udp or telnet then it will be possible to make a bridge to other programs (cw-skimmer, dx-cluster etc..)