cjcliffe/CubicSDR #798: Why are all people authorized to edit wiki pages?

Dantali0n, 2020-03-16:
Asked by @magentos-pl by creating wiki pages in Polish...

parkerlreed, 2020-03-16:
I mean isn't that part of the point of a wiki? Users can contribute updated documentation easier than a full pull request. 

I mean yeah if it gets out of hand and is abuse then a lockdown to contributors would be good.

vsonnier, 2020-04-12:
@parkerlreed Agreed that anybody should be able to contribute, but on the other hand nothing is stopping fucking up the whole thing... I don't see how to simply revert a whole commit, except by comparing 2 revisions and reverting line-by-line.

Dantali0n, 2020-04-12:
I get were the question by @magentos-pl is coming from, I have seen wikis defaced before and it is basically impossible to restore. Maybe editing wiki's by pull request is a better approach.

vsonnier, 2020-04-17:
@cjcliffe  Has the ability to control access to the Wiki if he wish to, in Settings > Restrict editing to collaborators only.

cjcliffe, 2020-06-17:
@vsonnier seems like a good idea; I've updated the settings

closed 2020-06-17 by cjcliffe