cjcliffe/CubicSDR #800: HamLib USB Backend not working because CubicSDR does not open the USB port?

g0oan, 2020-03-22:
Morning Everyone,

For a little while now I've been trying to get my Funkamateur FA-SDR kit controlled directly by CubicSDR from the radio menu, without success I'm afraid. 

The FA-SDR kit is included as part of the HamLib SDR/USB kit back end and while the rig does appear in the rig menu I've not been able to get CubicSDR talking directly, via the USB port to the rig.

As we all seem to a little time on our hands in London at the moment I thought I would try and compare the code from CubicSDR with WSJT-X which I know works perfectly with the FA-SDR kit. 

Now I'm not a programmer so I'm happy to be told I'm taking rubbish here but from what I can figure out the WSJT-X code calls a "int usb_port_open(hamlib_port_t *port)" function which appears to open the USB port and a connection to the rig. Apologies if I'm simplifying the language here.

I can't for the life of me find any similar function in the CubicSDR code. 

Would this be correct? Is the reason I can't get the CubicSDR / HamLIb USB backend working with my SDR kit is the CubicSDR doesn't open the USB port?

Once again apologies if I'm talking complete rubbish here I'm just curious.

Thanks in advance for any help.

Sean.

cjcliffe, 2020-06-17:
@g0oan I'm not familiar with the requirement to open the USB port as I'm using the higher level functions that should abstract the device itself.  As far as I know usb_port_open is an internal hamlib function that shouldn't be called directly.

With that said you don't appear to be the only one having an issue; it looks like this is the same problem as #757 and they appear to at least have a workaround by using the rigctld to bridge to the USB device.

PA0RWE, 2024-04-09:
Hi,
I'm using a Si570 SDR (KTH-SDR) and have the same problem with the USB connection. I did not understand if there is a solution for this problem. I hope, because I love CubicSDR!
Please help!
73's
Rob PA0RWE