cjcliffe/CubicSDR #805: v0.2.6a on HiDPI Screen

jawatson, 2020-04-05:
I've compiled CubicSDR v0.2.61 on a Dell XPS with a HiDPI screen (3840x2160).  The screen scaling is set to 200%. 

Everything appears to work correctly but the dialogue box used for entering frequencies manually is too small to display any content.

As a quick fix for my computer, I changed the dialogue size in /src/CubicSDR.cpp, line 974 as follows, setting the size to 700x150;

    FrequencyDialog fdialog(appframe, -1, title, activeModem, wxPoint(-100,-100), wxSize(700, 150), wxDEFAULT_DIALOG_STYLE, targetMode, initString);

This is obviously not a general fix but gets the application running on my machine.  The attached two photos show the dialogue box before and after the modification.
[image]
[image]

Dantali0n, 2020-04-06:
I think it would be better to continue discussing Retina / HiDPI problems in an already created issue #758.

jawatson, 2020-04-06:
Apologies - I didn't see the original issue when scanning through.  I've submitted a pull request for this particular issue.

vsonnier, 2020-04-10:
@jawatson  Thanks for

closed 2020-04-10 by vsonnier