cjcliffe/CubicSDR #807: Tips for wifi?

markdrayton247, 2020-04-07:
I'm attempting to stream from soapySdr server to my mac running Cubic. Has anyone got any experience of doing this over wifi ? The stream is pretty choppy. To be fair it was choppy over ethernet until I switched up the mtu to 4000 but I can't really do this with wifi....  can I throttle the bandwidth requirement somehow ? changing sample rate doesn't seem to help.
thanks

Dantali0n, 2020-04-15:
For starter you can try putting the UI of CubicSDR in slow mode. If any bandwidth limit would exist it will be in soapsdr remote not CubicSDR as it is much harder to implement a bandwidth limiter in a receiver as opposed to a tranmsitter.

vsonnier, 2021-09-24:
Issue is too old, no feedback, closing. Thanks @Dantali0n to have provided support !

closed 2021-09-24 by vsonnier