cjcliffe/CubicSDR #809: Installing CubicSDR on Pi4B

W9KJO, 2020-04-11:
When I run the Make command for CubicSDR  I get this

[100%] Linking CXX executable x86/CubicSDR
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::setFrequency(long long)':
DemodulatorPreThread.cpp:(.text+0x7e8): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::getFrequency()':
DemodulatorPreThread.cpp:(.text+0x810): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x81c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::setSampleRate(long long)':
DemodulatorPreThread.cpp:(.text+0x844): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::getSampleRate()':
DemodulatorPreThread.cpp:(.text+0x86c): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x878): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::run()':
DemodulatorPreThread.cpp:(.text+0x1f44): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x1f68): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x1f74): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x2040): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x205c): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x2098): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x2160): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o:DemodulatorPreThread.cpp:(.text+0x2af8): more undefined references to `__atomic_load_8' follow
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::run()':
DemodulatorPreThread.cpp:(.text+0x2e30): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x3aa4): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x3b50): undefined reference to `__atomic_load_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x3b64): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o: in function `DemodulatorPreThread::DemodulatorPreThread(DemodulatorInstance*)':
DemodulatorPreThread.cpp:(.text+0x4cb4): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x4cc8): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x4d0c): undefined reference to `__atomic_store_8'
/usr/bin/ld: DemodulatorPreThread.cpp:(.text+0x4d20): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::setFrequency(long long)':
CubicSDR.cpp:(.text+0xe6c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0xea8): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0xf40): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::getOffset()':
CubicSDR.cpp:(.text+0xf68): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::setOffset(long long)':
CubicSDR.cpp:(.text+0xf88): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0xfbc): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::getFrequency()':
CubicSDR.cpp:(.text+0x10a8): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::lockFrequency(long long)':
CubicSDR.cpp:(.text+0x10dc): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::setSampleRate(long long)':
CubicSDR.cpp:(.text+0x1198): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x11c0): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x1280): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::getSampleRate()':
CubicSDR.cpp:(.text+0x15c0): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::setDevice(SDRDeviceInfo*, int)':
CubicSDR.cpp:(.text+0x2d8c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2db8): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2dc8): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2dd8): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2e0c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2e28): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2f88): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x2fc4): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x3004): undefined reference to `__atomic_load_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x302c): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o: in function `CubicSDR::OnInit()':
CubicSDR.cpp:(.text+0x3980): undefined reference to `__atomic_store_8'
/usr/bin/ld: CubicSDR.cpp:(.text+0x3994): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o: in function `DeviceConfig::DeviceConfig()':
AppConfig.cpp:(.text+0x864): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o: in function `DeviceConfig::setOffset(long long)':
AppConfig.cpp:(.text+0x8c4): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o: in function `DeviceConfig::getOffset()':
[... 48 lines cut ...]
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x37b4): undefined reference to `__atomic_store_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x37c0): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x3828): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x39fc): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x3a14): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x3d78): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x3d90): undefined reference to `__atomic_store_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x3fa8): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SoapySDRThread.cpp.o: in function `SDRThread::init()':
SoapySDRThread.cpp:(.text+0x4410): undefined reference to `__atomic_store_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x4908): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x4aa4): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x4ab8): undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1353: x86/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

How do I fix this.  I have spend many hours trying to get this to work.

I followed all the steps in the Cubic SDR build list and ried the SDRPlay method.  Tried from a fresh Pi image.  Still nothing seems to work

Please help me.

Thanks. Walter

SDRplay, 2020-04-11:
Try this...

cmake .. -DOTHER_LIBRARIES="-latomic"

along with the other cmake options of course

W9KJO, 2020-04-11:
> 
> 
> Try this...
> 
> cmake .. -DOTHER_LIBRARIES="-latomic"
> 
> along with the other cmake options of course

Yes, That did it.  Thank you so much.  This has been a long couple of days.  But it is working now and looks great.

Thank you very much

Walter

cpedw, 2020-05-25:
I used Add/remove programs from the Preferences menu. After typing in SDR, I ticked one box and Apply. Job done!

Derek