cjcliffe/CubicSDR #811: Odd Graphics Resolution Issue

wcravens, 2020-04-12:
Hello, I've just put Cubic on a new Ubunti 18.04 installation for an RSPDuo.  It seems to be running fine but the graphics are oddly out of whack.  It seems that a lot of the renderings are half of what they should be.  And the mouse pointer is off too.  I'm including a screen-shot because it's hard to explain.

[image]

Dantali0n, 2020-04-12:
Seems related to https://github.com/cjcliffe/CubicSDR/pull/769

Do you happen to use display scaling as is typically done on 2k / 4k monitors?

vsonnier, 2020-04-19:
@Dantali0n  is right, it is a duplicate of #727.

closed 2020-04-19 by vsonnier