cjcliffe/CubicSDR #813: CubicSDR not finding Soapy module

4nanook, 2020-04-16:
I recently upgraded to Ubuntu 20.04 from 19.10.  In 19.10 I had to build CubicSDR because the version provided in the repositories segfaulted during an mmap() call.  But it was working.  When I tried to use my version after upgrading to 20.04 it would not start the SDR.  So I cleared my version and tried the version in the repository again, still segfaulted at the same place.  So I built it again following the build instructions in the Wiki to the letter.  Yet it tells me it can't find a soapy module.

cjcliffe, 2020-06-17:
Hi @4nanook -- I couldn't tell from your description, but did you also compile the necessary Soapy modules for your device?  They are separate from the CubicSDR build itself.

4nanook, 2020-06-17:
I had installed them from the Ubuntu distro.  I did later compile and
everything is working now.  I think the Ubuntu modules would not load into
my kernel because I'm running a self-compiled upstream kernel.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.