cjcliffe/CubicSDR #820: Unable to  operate Upper Sideband mode in CubicSDR vers 0.2.4

swlchris, 2020-05-13:
HI
I just got my Fedora 32 build  to  accept  the  new build of CubicSDR.
It finds the SOapySDR module just fine and everything works  mechanically with the SDRPlay RSP1A.
I can't use the UpperSideband mode. It  presents  zero audio except some crackling and snapping in the  headphones. In any other mode it seems to work fine.
Here are screenshots
first the  lower sideband operating
[image]
now the upper side
[image]
band operating
Is there something different I should be seeing?  any filter settings that could be  erasing the USB reception?
Thanks for looking.
Chris

vsonnier, 2020-05-13:
Hi there,  lots of things are not right with this example you gave us:

- The screenshot shows v0.26a, please correct your title. 
- Your WF do not look like you are listening to a USB station at all, but rather random noise so please pick a better example. 
- Zoom in on the main WF, this is especially not usefull as it is.
- **You are using Central Frequency = (current VFO) Frequency, so no wonder you hear only noise-like sounds because of the DC spike removal algorithm happening there.** Do as recommended by SDRPlay, shift your central frequency away from your VFOs.
- Look like you are using AGC. Do not use it and play with manual controls instead.
- List the settings you are using: RF select in particular ?

Last, do not use Low IF mode as it does not work as it is (and so has been "hidden" in the future 3.07 module version, see here: https://github.com/SDRplay/SoapySDRPlay ) so only use Zero-IF if you use the SoapySDRPlay current version.

Dantali0n, 2020-05-13:
> Hi there, lots of things are not right with this example you gave us:
> 
>     * The screenshot shows v0.26a, please correct your title.
> 
>     * Your WF do not look like you are listening to a USB station at all, but rather random noise so please pick a better example.
> 
>     * Zoom in on the main WF, this is especially not usefull as it is.
> 
>     * **You are using Central Frequency = Frequency  (current VFO), so no wonder you hear only noise-like sounds because of the DC spike removal algorithm happening there.** Do as recommended by SDRPlay, shift your central frequency away from your VFOs.
> 
>     * Look like you are using AGC. Do not use it and play with manual controls instead.
> 
>     * List the settings you are using: RF select in particular ?
> 
> 
> Last, do not use Low IF mode as it does not work as it is (and so has been "hidden" in the future 3.07 module version, see here: https://github.com/SDRplay/SoapySDRPlay ) so only use Zero-IF if you use the SoapySDRPlay current version.
> 
> There is liitle difference between USB and LSB algorithm, so if you can listen to one type, there is no reason you can't with the other. Both Works for Me (TM).

I want to add that the squelch while using USB is set so high that the output is muted so manually lowering the squelch might be all that is needed to fix this.

vsonnier, 2020-05-13:
> I want to add that the squelch while using USB is set so high that the output is muted so manually lowering the squelch might be all that is needed to fix this.

Indeed yes, I missed this !

swlchris, 2020-05-13:
Well I got it to work once. And of course, I can';t remember what I did to get it to work.  It reverted back to same old crap  when I exited and brought cubicsdr back up again
I went through all the suggestions  and double checked the center frequency since it likes to change  with tuning

This is what I get on LSB mode 
[image]

This is what I get on USB 

[image]

No audio at all on USB  there is audio on LSB

vsonnier, 2020-05-14:
OK something is going on here, in #818 @deltatango2 indeed reported the same thing using SDRPlay RSPDx :

> I can hear output in all other modes including LSB. When I switch to USB audio output disappears, I am not sure if I am missing something operationally with the software. I was not able to find in the docs or the UI any setting that would rectify it. The audio waveform on the top right also goes to zero in USB, so its not just a machine audio issue.
Thanks,

vsonnier, 2020-05-14:
@swlchris @deltatango2 please describe your CubicSDR context as best as you can:
- which Linux ? 32bit binary ? 64 bit ? Prebuilt ? Compiled from code ?...etc.
- liquid-dsp version ? 
- CubicSDR settings : Show us the Settings menu in a screenshot, that is enough.
- SoapySDRModule version ?
- Compiler and flags ? (GCC version...etc)
 
I own a RSP2 running a win64 CubicSDR on Windows10 and all I can say it is working for me. I don't think it is about being a particular device, but rather a DSP problem so it is pointing at liquid-dsp library issue. So what are the version you used ? ...etc.. 
Mine is compiled  from `liquid-dsp` 00e92db1db13e4c894e3c6ea108b23195d58f20d using a GCC 9.3 (MSYS2) but it is of small use if you compile on Linux.  

In short, any information could be useful.

@Dantali0n Thanks for participating ! I think you run on Linux too, so I assume it is working for you.

vsonnier, 2020-05-14:
Hem... let's see in the CubicSDR code:

- `ModemLSB.cpp(60)` : `firhilbf_c2r_execute(c2rFilt, x, &demodOutputData[i], &usb_discard);`
- `ModemUSB.cpp(60)` : `firhilbf_c2r_execute(c2rFilt, x, &lsb_discard, &demodOutputData[i]);`

Well well, unless C code as become incredibly polymorphic overnight, I would say the second call has its arguments mixed-up.

vsonnier, 2020-05-14:
DSP code calling liquid-dsp on CubicSDR look good, despite my first analysis. Having looked at liquid-dsp test examples, the differences between LSB and USB looks intentionnal, with C-isms all over the place. Damn, I __hate__ C. 

I'm going to have a closer look to liquid-dsp test examples when I have time. 

In the meantime @swlchris @deltatango2 please describe your CubicSDR context as best as you can:
- which Linux ? 32bit binary ? 64 bit ? Prebuilt ? Compiled from code ?...etc.
- liquid-dsp version ? 
- CubicSDR settings : Show us the Settings menu in a screenshot, that is enough.
- SoapySDRModule version ?
- Compiler and flags ? (GCC version...etc)
 
I own a RSP2 running a win64 CubicSDR on Windows10 and all I can say it is working for me. 

I'll have to test again with a win32 build.

I don't think it is about being a particular device, but rather a DSP problem so it is pointing at liquid-dsp library issue.  

Mine is compiled  from `liquid-dsp` 00e92db1db13e4c894e3c6ea108b23195d58f20d 
using a GCC 9.3 (MSYS2) but it is of small use if you compile on Linux.  

In short, any information could be useful.

@Dantali0n Thanks for participating ! I think you run on Linux too, so I assume it is working for you.

Dantali0n, 2020-05-14:
@vsonnier  I don't have the RSPDx though but a RSP 1, I have just tested this and everything worked fine both USB and LSB.

* CubicSDR - latest master
* SoapySDR - 0.7
* Liquid-DSP - 1.3.2 (commit 4892ebbc04ef57dd6f603d3f4ece253d8c2bd571)

Some other basic info on my system:
```bash
System:    Kernel: 5.3.0-46-generic x86_64 bits: 64 Desktop: i3 4.14.1
           Distro: Ubuntu 18.04.4 LTS
CPU:       Quad core Intel Core i5-4690K (-MCP-) cache: 6144 KB
           clock speeds: max: 4100 MHz 1: 1819 MHz 2: 1878 MHz 3: 1843 MHz 4: 1858 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X]
           Display Server: x11 (X.Org 1.20.5 ) driver: modesetting
           Resolution: 1920x1080@60.00hz, 1920x1080@60.00hz, 1920x1080@60.00hz
           OpenGL: renderer: AMD Radeon RX 470 Graphics (POLARIS10, DRM 3.36.0, 5.3.0-46-generic, LLVM 9.0.0)
           version: 4.5 Mesa 19.2.8
```

Dantali0n, 2020-05-14:
> @vsonnier I don't have the RSPDx though but a RSP 1, I have just tested this and everything worked fine both USB and LSB.
> 
>     * CubicSDR - latest master
> 
>     * SoapySDR - 0.7
> 
>     * Liquid-DSP - 1.3.2 (commit 4892ebbc04ef57dd6f603d3f4ece253d8c2bd571)
> 
> 
> Some other basic info on my system:
> 
> ```shell
> System:    Kernel: 5.3.0-46-generic x86_64 bits: 64 Desktop: i3 4.14.1
>            Distro: Ubuntu 18.04.4 LTS
> CPU:       Quad core Intel Core i5-4690K (-MCP-) cache: 6144 KB
>            clock speeds: max: 4100 MHz 1: 1819 MHz 2: 1878 MHz 3: 1843 MHz 4: 1858 MHz
> Graphics:  Card: Advanced Micro Devices [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X]
>            Display Server: x11 (X.Org 1.20.5 ) driver: modesetting
>            Resolution: 1920x1080@60.00hz, 1920x1080@60.00hz, 1920x1080@60.00hz
>            OpenGL: renderer: AMD Radeon RX 470 Graphics (POLARIS10, DRM 3.36.0, 5.3.0-46-generic, LLVM 9.0.0)
>            version: 4.5 Mesa 19.2.8
> ```

PSA: I had to blacklist a module which I haven't seen before it was called msi2500 and kept claiming the SDRPlay RSP1 device.

deltatango2, 2020-05-14:
@vsonnier sorry for the delay.  I'm away from my linux box atm, but I was running on the latest raspbian on an Rpi4, compiled all with the latest release of each of the dependent packages.  Once I get to my machine I can post the hashes.  The soapysdr module for the latest RSP API 3.07 requires the package direct from SDRplay vice pothos, so I wonder if the issue is there.

Dantali0n, 2020-05-14:
> @vsonnier sorry for the delay. I'm away from my linux box atm, but I was running on the latest raspbian on an Rpi4, compiled all with the latest release of each of the dependent packages -O2 flag. Once I get to my machine I can post the hashes. The soapysdr module for the latest RSP API 3.07 requires the package direct from SDRplay vice pothos, so I wonder if the issue is there.

Since you are running on an Raspberry Pi I would also be very interested in the console output of CubicSDR while it is running. I greatly suspect the Rpi not being able to keep up and discarding sample data as I have seen this happen before.

swlchris, 2020-05-14:
Hi
I am working at the moment so will get that info out after work.
My system is on Fedora 32 x64
Have to check on other info
Thanks
Chris

swlchris, 2020-05-14:
WEll guess what? The whole CubicSDR took a dump on me. Now it doesn't want to work at all in any mode. NO waterfall no signals no nothing. I upgraded from Fedora 31 to  Fedora 32 and followed the instructions to install from the github page, but I'm thinking there might have been residual files left over from previous  installs on F31 for  SOapySDR CUbicSDR and  other  controllers that  may have blown this up.
So, I"m going to go a complete fresh reinstall of F32 and go from there.I will use the Fedora 32  page here
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux
Hopefully this  will resolve the issue... 
Thanks for the insights and  help .. will see how we do here

swlchris, 2020-05-17:
Hi
Ok, I  did a complete reinstall of Fedora32 64 bit. from CD not an upgrade from  previous Fedora.
I followed  this guideline ..
https://soundbytes.asia/proppy/blog/page/51/sdrplay-cubicsdr-on-fedora-32/
I had to install a few things of course because Fedora on DVD is 2 gb and nothing is included regarding development tools.
I got every thing installed and when to run CubicSDR and the  message came up 
CubicSDR: error while loading shared libraries: libliquid.so: cannot open shared object file: No such file or directory
This is after the installation of liquid-dsp was declared a success by the  manual installer following the  instructions.
 So a complete hour is wasted  in installing and  doing a step by step installation.
The only way I got my SDRPlay RSP1A to play is to use a  generic binary of  GQRX at the moment and it found the SDRSoapy module.So at least I know the  sdrplay runtime worked.
So then I  researched the liquiddsp issue and found this from  a  forum somewhere.
https://github.com/windytan/redsea/issues/50 
It's about liquid-dsp.so.xx not showing up and I followed that  tip  and did the export as shown.
export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
Then I tried Cubic SDR again and this time got the 
CubicSDR: error while loading shared libraries: libSoapySDR.so.0.8: cannot open shared object file: No such file or directory
So I did this as shown before  as an issue from https://github.com/pothosware/SoapySDR/issues/105
export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
 and now CubicSDR works and the USB is also working right out of the box.
 Now the only problem is. I have to do this each time I restart the computer for CubicSDR to work.The liquid-dsp errors come up and the libsoapysdr error continue. Once I went back did the two different exports. it  comes up running CubicSDR in terminal mode.
 Any guesses as to what I can do to fix this? NOt exactly a Linux genius here.
 The  squelch and the  settings are shown  here 
 Hi
Ok, I  did a complete reinstall of Fedora32 64 bit. from CD not an upgrade from  previous Fedora.
I followed  this guideline ..
https://soundbytes.asia/proppy/blog/page/51/sdrplay-cubicsdr-on-fedora-32/
I had to install a few things of course because Fedora on DVD is 2 gb and nothing is included regarding development tools.
I got every thing installed and when to run CubicSDR and the  message came up 
CubicSDR: error while loading shared libraries: libliquid.so: cannot open shared object file: No such file or directory
This is after the installation of liquid-dsp was declared a success by the  manual installer following the  instructions.
 So a complete hour is wasted  in installing and  doing a step by step installation.

The only way I got my SDRPlay RSP1A to play is to use a  generic binary of  GQRX at the moment and it found the SDRSoapy module.So at least I know the  sdrplay runtime worked.

So then I  researched the liquiddsp issue and found this from  a  forum somewhere.
https://github.com/windytan/redsea/issues/50 
It's about liquid-dsp.so.xx not showing up and I followed that  tip  and did the export as shown.
export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
Then I tried Cubic SDR again and this time got the 
CubicSDR: error while loading shared libraries: libSoapySDR.so.0.8: cannot open shared object file: No such file or directory
So I did this as shown before  as an issue from https://github.com/pothosware/SoapySDR/issues/105
export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
 and now CubicSDR works and the USB is also working right out of the box.

 Now the only problem is. I have to do this each time I restart the computer for CubicSDR to work.The liquid-dsp errors come up and the libsoapysdr error continue. Once I went back did the two different exports. it  comes up running CubicSDR in terminal mode.

 Any guesses as to what I can do to fix this? Not exactly a Linux genius here.
 The  squelch and the  settings here 
[image]
 I can't seem to get a screen shot of the settings.
Zero IF is used
AGC  auto is off
RF gain at 0
IQ correction on.

Fedora 32 x64_86
SDRplay module 
SDRplay_RSP_API-Linux-2.13.1run
CubicSDR v0.2.6a
up to date GCC cmake and the development tools.
Intel CPU
Thanks
Chris

vsonnier, 2020-05-17:
> and now CubicSDR works and the USB is also working right out of the box.

Great, so the problem was likely in a intermediate `liquid-dsp` version exposing a particular bug.

> Now the only problem is. I have to do this each time I restart the computer for CubicSDR to work.The liquid-dsp errors come up and the libsoapysdr error continue. Once I went back did the two different exports. it comes up running CubicSDR in terminal mode.

At that point, simply put them in a script `CubicSDR.sh` to have them set each time you run CubicSDR:

```bash
#! /bin/bash
export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
./CubicSDR
```
Another alternative is to put those 2 exports in your `~/.bashrc` but it will affect all applications. 

> Not exactly a Linux genius here.

Me neither !

Dantali0n, 2020-05-17:
> export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
> Then I tried Cubic SDR again and this time got the
> CubicSDR: error while loading shared libraries: libSoapySDR.so.0.8: cannot open shared object file: No such file or directory
> So I did this as shown before as an issue from [pothosware/SoapySDR#105](https://github.com/pothosware/SoapySDR/issues/105)
> export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
> and now CubicSDR works and the USB is also working right out of the box.

Interesting, this is what the `sudo ldconfig` in the build instructions is for it should update these paths based on a file in **/etc/ld.so.conf.d**. Can you check if there is a file related to liquid-dsp in that directory. If it does not exist you could always create one manually they are just text files with an absolute path to the **lib** folder of the library / program you want to expose. Don't forget to run `sudo ldconfig` if you do place a new file there.

swlchris, 2020-05-17:
> Interesting, this is what the `sudo ldconfig` in the build instructions is for it should update these paths based on a file in **/etc/ld.so.conf.d**. Can you check if there is a file related to [...]

I looked and only two files are in there .
dyninst-x86_64.conf
libiscsi-x86_64.conf
 So to do as you suggested I would have to place a text file with the export command  in this directory relating to the  liquid-dsp and make it executable?
I  haven't had a chance to   experiment with  command structure  since I've  been more of a user so this is a learning experience.

Dantali0n, 2020-05-17:
> to place a text file with the export command in this directory relating to the liquid-dsp and make it executable?

Almost not quite, unlike `.bashrc` this file does should **not** contain any shell code **nor** be executable.
It should just contain the absolute path to the `lib` folder of where you installed liquid-dsp.

An example would be `/etc/ld.so.conf.d/liquid.conf` containing:
```
/opt/liquid-dsp/lib
```

swlchris, 2020-05-17:
> Almost not quite, unlike `.bashrc` this file does should **not** contain any shell code **nor** be executable.
> It should just contain the absolute path to the `lib` folder of where you installed [...]

HI
Progress!!
I put this  file in the /etc/ld.so.conf.d folder 
/home/chris/Dev/liquid-dsp/
 and CubicSDR came up but now it  can't find the soapySDR module
but if I got back to that script I created  with 
/bin/bash
export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
/home/chris/Dev/CubicSDR/build/x64/CubicSDR
then it comes right up.
I did name the file libSoapySDR.conf 
Would that be an issue? NO because I just  renamed it to SoapySDR.conf and I get the same error  running CubicSDR from terminal. it can't find the SoapySDR module it just used previously after that  /bin/bash script

The contents of the SoapySDR.conf is 
/home/chris/Dev/SoapySDR/build/lib/

Something isn't right  because I have a lot of choices when I  i  do the locate for the  libSoapySDR
Results
[chris@localhost ~]$ sudo locate libSoapySDR
/home/chris/Dev/SoapySDR/build/lib/libSoapySDR.so
/home/chris/Dev/SoapySDR/build/lib/libSoapySDR.so.0.8
/home/chris/Dev/SoapySDR/build/lib/libSoapySDR.so.0.8.0
/home/chris/Downloads/gqrx-sdr-2.11.5-linux-x64/lib/libSoapySDR.so
/home/chris/Downloads/gqrx-sdr-2.11.5-linux-x64/lib/libSoapySDR.so.0.6
/home/chris/Downloads/gqrx-sdr-2.11.5-linux-x64/lib/libSoapySDR.so.0.6.1
/usr/lib64/libSoapySDR.so.0.7
/usr/lib64/libSoapySDR.so.0.7.1
/usr/local/lib64/libSoapySDR.so
/usr/local/lib64/libSoapySDR.so.0.8
/usr/local/lib64/libSoapySDR.so.0.8.0
[chris@localhost ~]$ 

So going back to that CentOS  post about the export issue  I  tried the  following quoted fcrom that post 
"To make it permanent I created file sudo vim /etc/ld.so.conf.d/usrlocal.conf with contents
/usr/local/lib/
/usr/local/lib64/
then sudo ldconfig and everything is good."
It worked for me too.
So now my SoapySDR.conf file in the /etc/ld.so.conf.d  folder reads

/usr/local/lib64

So apparently  the library gets lost somehow because the install placed it into the  lib64  are and not a usual library.
I'm going to reboot and see if it works  continuously
Thanks
Chris

swlchris, 2020-05-17:
So apparently the library gets lost somehow because the install placed it into the lib64 are and not a usual library.
I'm going to reboot and see if it works continuously
Thanks
Chris

Well that did the trick.
So now it  shows up when I click on the icon on Gnome desktop which it didn't do before due to all the errors I'm sure. The USB works  evenly with the LSB and AM modes. I'm keeping the center frequency away from listening frequency and  the squelch is set  low and the agc is off and manually  controlling gain.
Thanks for all the help  it's appreciated... I would hate to have to  use Windows  to listen to my SDR on a regular basis when I just don't feel like  using it as a panadapter for the TS870
Chris

deltatango2, 2020-05-27:
> Since you are running on an Raspberry Pi I would also be very interested in the console output of CubicSDR while it is running. I greatly suspect the Rpi not being able to keep up and discarding [...]

Sorry for the delay.  Back up and running on the Pi.  Unfortunatley, the problem has rectified itself and I don't have a good reason why.

On a different note, while the unit is running, I do not see any errors printing to the console and CPU usage is 31.8% and mem 13.6%.  Sorry I can't contribute anything useful to the resolution.

nomadiq, 2020-11-04:
I think I fixed this problem. There was an error in the CubicSDR/src/modules/modem/analog/ModemUSB.cpp file. I fixed the order of the arguments in the firhilbf_c2r_execute function, recompiled and USB now works fine in v0.2.6a. I created a pull request.

matt-domsch-sp, 2020-11-10:
I have packages for Fedora 31 and 32 available here: https://copr.fedorainfracloud.org/coprs/mdomsch/CubicSDR/

Fedora 33 has a fairly recent CubicSDR, liquid-dsp, and Soapy bits, and I just built and will push latest CubicSDR as of August 24 to Fedora 33 updates-testing and rawhide repos.

deltatango2, 2021-03-15:
@nomadiq I applied your patch to my Ubuntu x64 install and it corrected the USB no audio issue.  Thanks

vsonnier, 2022-01-15:
As I stated elsewhere (don't find the issue) @nomadiq patch is wrong, and the actual code is right:

- ModemLSB.cpp : ` firhilbf_c2r_execute(c2rFilt, x, &demodOutputData[i], &usb_discard);`
- ModemUSB.cpp : `firhilbf_c2r_execute(c2rFilt, x, &lsb_discard, &demodOutputData[i]);` 

i.e. You discard the other part you are not interested in, and the arguments are lsb then usb. 

Anyway, the issue is too old, closing.

closed 2022-01-15 by vsonnier

nomadiq, 2022-01-16:
Well whatever, my fix fixed the issue I had in 0.2.6a and others had seen as well. I dunno what the problem is.