cjcliffe/CubicSDR #821: "real" scanner mode

HyperDevil, 2020-05-13:
Hi,
I am looking for an enhancement where i can cycle through my bookmarks and listen to channels with activity. (level above squelch)
This to simulate a real hardware scanner, like a Uniden or AOR.
Its a feature i think no other SDR has, if someone adds this to cubicSDR it will be top-of-the-line.
Not sure about the speed (and what the CPU can handle) so some sort of speed toggle and lockout feature is probably a good idea.

Dantali0n, 2020-05-14:
The primary bottleneck in this is not likely to be the CPU but rather the speed at which the SDR can lock to a new desired frequency. As almost every consumer SDR has only one receiver channel it will only be able to demodulate a portion of bandwidth at one frequency at a time (the width of this portion being a function of the sampling rate). The downside of implementing such a scanner would be that it would have to freeze the main window and waterfall while scanning through the configured frequencies that would otherwise be out of band.

Although I like the idea and think it can be a great feature we need to be a bit careful with this on some devices, particularly, the devices that contain hardware filters and switch these in and out of circuit using relays. As rapidly cycling relays can greatly reduce their lifespan.

HyperDevil, 2020-05-14:
Yes i agree with your opinion about the SDR/radio being the bottleneck when it comes to speed. For me personally the waterfall is not necessary when in scanning mode, since i already know the frequencies i want to listen too (from the bookmars), so for me that part of the GUI you can skip completely. What the most important thing is is the squelch, since different channels have different noise levels (in the future maybe a squelch per channel). Yes the lifespan thing can be a concern on some devices, although some atm are so cheap (1/10 of the price of a uniden scanner), giving the user a warning about this should be OK in my opinion. The scanning function is so sawed after, but no SDR software kit has it.

SDRplay, 2020-05-14:
I don't want to interrupt this discussion on scanning, but it is not true to say that no SDR software has it.

HyperDevil, 2020-05-14:
Hi SDRplay, can you show me any software that has this true scanning functionality ?

SDRplay, 2020-05-14:
haha - I like the additional bit you added at the end :-)

Dantali0n, 2020-05-16:
> Hi SDRplay, can you show me any software that has this true scanning functionality ? (which is compatiable with RTL SDR or HackRF)

I think this application could provide the features you request, however, it only does AM demodulation: https://github.com/szpajder/RTLSDR-Airband

SDRplay, 2020-05-16:
@Dantali0n that's why I laughed in my last comment. It isn't true to say that there isn't any SDR software that can do scanning - and then the bit about RTL or HackRF was added in an edit.

I wanted to correct the statement without hijacking a CubicSDR issues thread by starting to talk about other SDR software explicitly. You can probably have a guess as to what software I'm talking about anyway.

HyperDevil, 2020-05-16:
Would it be possible to incorporate the code from RTLSDR-airband into cubicsdr? (re-use functions?)
I was not aware of any supported piece of software with scanning functionality for the radios cubic supports (hackrf which i use).

MCKLtech, 2020-05-18:
SDR# has a scanning module available. Yes, I know it's Windows, but it could be a good template for what can be achieved.

ankostis, 2020-12-01:
Thiere are some screenshots from the SDR# scanner plugin: https://www.rtl-sdr.com/sdrsharp-plugins/