cjcliffe/CubicSDR #822: how to enable jack audio output

juankr10s, 2020-05-14:
hi ..
how to compile cubicsdr to enable jack output?
only show pulseaudio output 
i need send i/o mode cubicsdr to baudline via jack audio connection

SDRplay, 2020-05-28:
Not sure if you are using RPi or not, but the combination of CubicSDR and PulseAudio on RPi just doesn't work for me AT ALL - terrible problems, however...

when building our latest image, I built CubicSDR with ALSA audio support instead, now it's all working beautifully (HDMI, 3.5mm jack & USB audio)

Note: Only RPi tested, not sure on other hardware. Hope that helps.

Andy

juankr10s, 2020-05-30:
im using ubuntu 18.04 notebook hp
it work with pulseaudio only max is 96khz bw
im set pulseaudio in 24bit and 192khz but only cubicsdr work show menu 96khz

calimeroteknik, 2024-12-13:
Sneaky! I expected the RtAudio backend meant ALSA and Jack support, but it didn't work…
Why are the backends disabled in CMakeLists.txt (emphasis mine)?
>SET(USE_AUDIO_PULSE ON CACHE BOOL "Use Pulse Audio")
>SET(USE_AUDIO_JACK ***OFF*** CACHE BOOL "Use Jack Audio")
>SET(USE_AUDIO_ALSA ***OFF*** CACHE BOOL "Use ALSA Audio")
>SET(USE_AUDIO_OSS ***OFF*** CACHE BOOL "Use OSS Audio")

Lots of time lost finding out why it didn't work out of the box. Would be nice to have them on by default!