cjcliffe/CubicSDR #826: auto-gain behaviour coming from somewhere

fastfourier666, 2020-05-28:
Hey, I couldn't find anywhere else to post this - it doesn't seem much like an "issue" but there you go

I'm decoding some RF remotes using SDRplay (over a network). I have turned off the auto-gain in the File menu, but after the remote sends the data, the background noise creeps back up. The closest I could find in the manual is the "Audio Gain" slider, but that just makes the whole thing louder or quieter. 

I made a video: (warning: the signal is loud!) [https://studio.youtube.com/video/HOnHD2KlN7Q/edit](url)

Is this something on the SDR or something in the modem? I'm thinking the latter as the waterfall stays steady, but I am really new to this! Thanks!

cjcliffe, 2020-06-16:
Hi @fastfourier666 

This is default behavior of the AM/LSB/USB modems as I designed it; it will attempt to automatically gain the audio as the level rises or drops.

I need to double check -- but I believe if you set the signal squelch slider to be above the noise floor it should automatically lock the audio gain where it was until squelch is broken again.

If you see a need to set a fixed audio gain for the AM/LSB/USB modem it should be possible to add an option for it; let me know and I'll convert this to a feature request.

As far as I know it's only this way since I wanted automatic adjustment of the volume level while listening to AM / SSB as it varied widely while listening to a single conversation consisiting of different signal levels from different locations or navigating around to different signals.

zezic, 2020-09-01:
I would like to have an option to lock the modem gain too.

NeuroForLunch, 2021-05-09:
Add me to the list of people wanting an option to lock the modem gain

colbyAtCRI, 2021-05-20:
I've added a ModemAnalogVC which has ModemAnalog as a base class. Basically all analog modems that use this base class inherit a user settable modem gain. One may switch between automatic gain (as it currently is done) or select a discreet set 0,5, 10, 15, 20...,60 dB gain. 

[image]

NeuroForLunch, 2021-05-20:
@colbyAtCRI Please make the pull request

colbyAtCRI, 2021-05-20:
I’ve made the pull request.