cjcliffe/CubicSDR #827: CubicSDR hangs on startup with Hamlib support compiled

giordyb, 2020-05-29:
Hello everyone,

I'm running CubicSDR on a Raspberry PI 4 with 4GB of RAM and a SDRPlay RSP1A with the latest Raspbian and so far it's running great. 
CubicSDR is compiled from source with the following configuration

`cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config -DOTHER_LIBRARIES="-latomic" -DUSE_HAMLIB=0 -DUSE_AUDIO_PULSE=1
`

I wanted to add the HAMLIB integration to control CubicSDR from WSJT-X so downloaded and compiled and installed hamlib 3.3 from source and re-compiled CubicSDR with -DUSE_HAMLIB=1.

The build ran fine but when I start up CubicSDR it finds the SDR device but hangs at  "Loading SoapySDR modules...". and the list is blank. Sometimes it even shows the SDR device in the list but won't let me click on it in order to open the configuration and start it.

this is the output when it happens:

```
Loading:: configuration file '/home/pi/.CubicSDR/config.xml'
Loaded 230 rig models via hamlib.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.8.0
	ABI Version: v0.8
	Install root: /usr/local
	Loading modules... 
	Available factories...null, remote, sdrplay
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 19031BF996
  serial = 19031BF996
Make device 0
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: Yes
	DRI2SwapInterval: No
	glxSwapIntervalMESA: Yes
	glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono36_0.png', parsed 255 characters.
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
  serial=19031BF996

Reporting enumeration complete.
SDR enumerator done.
```

Other times I can click on the device but then i get this error repeating on the console.
```

SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...

```

If I recompile with hamlib disabled then everything goes back to normal.
Is anybody else having this issue?

cjcliffe, 2020-06-16:
@giordyb a couple of things to try that might narrow it down:

- Install SoapyAudio module (don't enable hamlib within SoapyAudio itself though) and try it without SDRPlay attached
- If that works start CubicSDR with SDRPlay unplugged, then plug it in, then refresh the list and try to start SDRPlay
- Try using an older version of hamlib and see if there's any difference

My initial guess is there might be some sort of lib conflict between what hamlib and SDRPlay is doing for USB access to devices.

giordyb, 2020-06-21:
Hi, thanks for the reply, I tried the steps you suggested:

- Installed SoapyAudio and recompiled CubicSDR with the Hamlib support compiled, without plugging in the RSP1a it works.

- Started CubicSDR with the RSP1a unplugged, plugged it in, scanned the list and after selecting the device CubicSDR hangs.
- Repeated the steps with Hamlib 3.3, 3.2 and 3.1 installed, same issue.

ve9gj, 2020-10-28:
Any progress on this issue?
I am finding the same problem after compiling with hamlib 4.0 from git
I started with pi image from the SDRplay website, which worked fine.  Built hamlib 4.0 from git. did a cmake .. with USE HAMLIB, make , sudo make install in /home/pi/Dev/cusbicsdr/build and now CubicSDR hang on the  "Loading SoapySDR modules...".  If the window is dismissed CubicSDR is still responsive.

73
Glenn VE9GJ