cjcliffe/CubicSDR #829: Device list empty ...

rszemeti, 2020-06-07:
Win7 64 bit, latest release of Cubic SDR, latest Zadig (2.5)

Nooelec Smart SDR dongle.

Follow the Zadig guide, do the reinstall driver thing on both interfaces,  driver installs with SUCCESS at bottom of Zadig ... and ... when I run Cubic, the available device list is empty ... just the headings: local, local net, remote etc ... not even any sound cards or default device is listed .. nothing. 

These are essentially newish plain vanilla windows installs.

I have 2 machines that do this ... the same software on my much used and stuffed with software dev laptop runs fine, but I can't keep dragging that around the shack.

Any clues on how to debug this?

Dantali0n, 2020-06-08:
> Any clues on how to debug this?

A first step would be to open the application from powershell / commanprompt so the debug output is visible. This will show the drivers SoapySDR was able to detect among other things.

rszemeti, 2020-06-08:
This is either incorrect or incomplete information.  when run from the command prompt there is no debug information available, nor is there a command line switch to enable debug output.

rszemeti, 2020-06-08:
[image]

See, no debug information available :(

Dantali0n, 2020-06-09:
> This is either incorrect or incomplete information. when run from the command prompt there is no debug information available, nor is there a command line switch to enable debug output.

I think you would need to compile it from source using `RelWithDebInfo`, I do remember having console output on Windows but have not used that OS in 5 years. If this does not work than running it from visual studio directly should definitely show the console output.

rszemeti, 2020-06-09:
That is way WAY beyond where I would want to go.  Just a simple user trying to get it to work. I tried it on yet another Windows 7 machine today,  and again, same story, so that is 3 machines now behaving the same way. If you ever get it fixed, please update this ticket. Pity, it looked really good, was looking forward to the rig control interface.

vsonnier, 2020-06-09:
@rszemeti Try to run with SDRSharp, to check if the fault is not your dongle or Zadig installation. For reference, I exclusivly use Windows and it works perfectly with the RTL-SDR. The tricky part is indeed installing Zadig properly. As SDRSharp use Zadig too, it is worth checking.

rszemeti, 2020-06-09:
I can try it, but the dongle works just fine under Windows on a 4th machine (my dev laptop) running Win7 64, but I don't wan't to use this in the shack. The other 3 computers all fail.   I will try SDRSharp on the problem machine and let you know the result.

vsonnier, 2020-06-09:
> [image]
> 
> See, no debug information available :(

In fact, your are not too far from getting one, use the following command:
```bash
CubicSDR.exe > startup.log
```
It will create a `startup.log` file where all the traces go. Attach the file contents to this issue. It also works with a regular command prompt too. 
The simplest way to run it at the right place is to type `cmd` in the Explorer address bar where Cubic is, it directly open a command prompt at the right place.

rszemeti, 2020-06-10:
Ah ha! I foolishly expected to see STDOUT and STDERR on the screen. Silly me.   OK,  Here you go:

Loading:: configuration file 'C:\Users\tom\AppData\Roaming\CubicSDR\config.xml'
Loaded 208 rig models via hamlib.
SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: C:\Program Files
	Loading modules... 
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//airspySupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//audioSupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//bladeRFSupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//HackRFSupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//RedPitaya.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//remoteSupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//rtlsdrSupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//sdrPlaySupport.dll..
Loading bundled SoapySDR module C:\Program Files\CubicSDR/modules//uhdSupport.dll..
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

	Available factories...airspy, audio, redpitaya, remote, uhd
Loaded font 'Bitstream Vera Sans Mono' from 'C:\Program Files\CubicSDR\fonts\vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'C:\Program Files\CubicSDR\fonts\vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'C:\Program Files\CubicSDR\fonts\vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'C:\Program Files\CubicSDR\fonts\vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'C:\Program Files\CubicSDR\fonts\vera_sans_mono27_0.png', parsed 255 characters.

Reporting enumeration complete.
SDR enumerator done.
Terminating SDR thread..

And the screenshot:

[image]

vsonnier, 2020-06-11:
@rszemeti Your `rtlsdrSupport.dll` is not loading at all, and so your `audioSupport` which in the other end should _always_ work if you have a sound card/chip installed. 

Try this version of CubicSDR instead (win64, VS2019). Unzip it somewhere, preferably not in `Program Files` nor in any other directory with spaces in its path: 
[CubicSDR_master_win64_20200606.zip](https://github.com/cjcliffe/CubicSDR/files/4763115/CubicSDR_master_win64_20200606.zip)

Also try SDRSharp.

rszemeti, 2020-06-12:
OK, I have tried a few times now with the VS2019 version you linked in a Zip ... it seems there are elements of the program compiled with much older versions of MSVC ... I know I have the 2015 MSVC runtimes,  this was complaining about MSVC120.dll .. which is the 2013 MSVC .. I installed that, now it is complaining about MSVC140.dll ... I'll try installing that in the morning, but it looks like it is not a clean compile against the current version of MSVC ...

vsonnier, 2020-06-12:
> it looks like it is not a clean compile against the current version of MSVC ...

Yes it is, but damn the fault is on some of the `modules`. I forgot that because I don't recompile all of them given the hassle, only copy existing ones.  You can remove the ones related to the devices you are not intrested in from this directory. The `rtl-sdr` module is VS2015 in particular.

I'm still waiting for you to try SDRSharp, this should be simple enough and a quick check to know if you have installed and more importantly configured Zadig drivers properly, following this guide : https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

rszemeti, 2020-06-12:
SDR Sharp will not load for me because it uses a later version of .NET than
I have installed, and this computer is in my remote shack and not connected
to the internet, so each test requires some walking ... I have not worked
out how to install .NET updates offline.

I will try again without the not-needed modules.

-- 

Best regards,

Robin Szemeti

Redpoint Consulting Limited

E: robin@redpoint.org.uk
T: +44 (0) 1299 405028
M: +44 (0) 7971 883371

CONFIDENTIALITY NOTICE
The information contained in this e-mail is intended only for the
confidential use of the above named recipient. If you are not the
intended recipient or person responsible for delivering it to the
intended recipient, you have received this communication in error
and must not distribute or copy it.
Please accept the sender's apologies, notify the sender immediately
by return e-mail and delete this communication.

Thank you.

Dantali0n, 2020-06-12:
> SDR Sharp will not load for me because it uses a later version of .NET than I have installed, and this computer is in my remote shack and not connected to the internet, so each test requires some [...]

Search for .NET offline installer for your required version. Here is the link to 4.8 as an example https://support.microsoft.com/en-us/help/4503548/microsoft-net-framework-4-8-offline-installer-for-windows

rszemeti, 2020-06-12:
OK, so I could not get the latest .ZIP you sent to run, even when I fixed the missing ,DLL's ... it exits without writing anything to the logfile.

However ... somewhere in loading the various MSVC runtimes, it has fixed the original install, which can now see and use the dongle, hurrah!

So I guess the problem is the release needs a variety of different runtimes, but only one is being installed/checked for ...

teroclro, 2021-01-08:
Hi
I have the smartee version dongle running win 10 and it too fails to list it
I have run and used SDR# ok proving the dongle is seen by the system and working but still not seen in cubicsdr

teroclro, 2021-01-14:
I have the same  problem sdr device not seen by cubicsdr 
Works OK with sdr#
Tried most options nooelec was no help

vsonnier, 2022-01-15:
Issue is too old, time to cleanup.

closed 2022-01-15 by vsonnier