cjcliffe/CubicSDR #830: CubicSDR git master version not working on RPi with Raspbian OS Buster

BuddyButterfly, 2020-06-08:
Hardware:
Raspberry PI 3

OS:
RPi OS Buster (current as of today) version "Raspberry Pi OS (32-bit) with desktop" (without additionl software). This OS uses LXDE.

CubicSDR:
- Curent git master version
- Compiled with all "standard libraries"  including wxWidges 3.1.3 on pi.

Symptom:
Cubic SDR does not show any graphics. Only small windows on the left side are show.  Functionality seems to be OK. It detects SDRPlay and it can be selected. See 
screenshot:

[image]

Counter-Verification:
Same build has been done on notebook. CubicSDR works perfect.

Any Ideas?

BuddyButterfly, 2020-06-08:
Reason found. On a direct display, CubcSDR shows up like it should.

So there is a problem when connecting remote via x2go or xrdp. In this mode, graphics is  not shown. I will further investigate.

Dantali0n, 2020-06-09:
> Reason found. On a direct display, CubcSDR shows up like it should.
> 
> So there is a problem when connecting remote via x2go or xrdp. In this mode, graphics is not shown. I will further investigate.

CubicSDR uses graphics acceleration using OpenGL which does not work over x2go / xrdp. afaik this limitation can not be resolved using x windows based remote desktop solutions.

cjcliffe, 2020-06-16:
@BuddyButterfly thanks for the submission, but it looks like supporting display protocols is outside the scope of CubicSDR itself.   

If you find something that works please feel free to follow-up as I'm sure others would be interested.

closed 2020-06-16 by cjcliffe