cjcliffe/CubicSDR #838: Build wxWidgets 3.1.3 under High Sierra, tiff.h not found

paulabama, 2020-07-09:
One small change to installing wxWidgets. Config file instructions should now read:

wxWidgets-3.1.3$ ./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff**=builtin** --with-libpng --with-zlib --with-mac --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --with-macosx-version-min=10.9 --prefix=/Users/(YOUR_USERNAME)/Dev/wxWidgets-staticlib CXXFLAGS="-std=c++0x" --with-libiconv=/usr

Please let me know if you have any questions.
spectrala.llc@gmail.com

vsonnier, 2020-12-25:
Please use the new procedure applicable to wxWidgets v3.14. (see Wiki)

closed 2020-12-25 by vsonnier

philippjbauer, 2022-07-01:
Ran into this issue with wxWidgets 3.15 on macOS 12.4 ARM64. Modifying the `--with-libtiff` option to `--with-libtiff=builtin` resolved the issue for me.