cjcliffe/CubicSDR #845: Layout on OSX Catalina

robhardt, 2020-08-31:
All of the spectrum displays & the waterfall appear to be scaled down to 1/4 the size of their enclosing panels.  See the attached screenshot.  I've tried multiple combos of CubicSDR and wxWidgets with the same results.  I'm just getting started with the codebase.  Any guidance as to where I can start poking around?  Thanks!

[image]

Dantali0n, 2020-08-31:
This is a duplicate of https://github.com/cjcliffe/CubicSDR/issues/758 we have a merge request open on https://github.com/cjcliffe/CubicSDR/pull/769 you could give this patch a go to see if it resolves your issues.

vsonnier, 2020-08-31:
Thanks @Dantali0n indeed it is a duplicate of #758. Please @robhardt continue the discussion there.

closed 2020-08-31 by vsonnier