cjcliffe/CubicSDR #847: Build failing with LIQUID_LIBRARIES error

TJNII, 2020-09-04:
Good evening:

I'm attempting to rebuild my CubicSDR Docker container and the compile is failing with the following error:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIQUID_LIBRARIES
    linked by target "CubicSDR" in directory /src/CubicSDR
```

The change date on https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux is several months before my last successful build on 2020-06-03 so I don't believe I missed a documentation update, but please let me know if there was a breaking change I need to update my build to account for.

Here is an exerpt of my Dockerfile with the build steps, the last build step posted is the one failing:

```
RUN apt-get install -y git build-essential automake cmake libgtk-3-dev freeglut3 freeglut3-dev
RUN apt-get install -y ca-certificates wget

RUN mkdir /src
WORKDIR /src

# Build and install SoapySDR
RUN git clone https://github.com/pothosware/SoapySDR.git
RUN cd SoapySDR && \
    mkdir build && \
    cd build && \
    cmake ../ -DCMAKE_BUILD_TYPE=Release && \
    make -j4 && \
    make install && \
    ldconfig && \
    SoapySDRUtil --info #test SoapySDR install

#Build and install liquid-dsp
RUN git clone https://github.com/jgaeddert/liquid-dsp
RUN cd liquid-dsp && \
    ./bootstrap.sh && \
    CFLAGS="-march=native -O3" ./configure --enable-fftoverride && \
    make -j4 && \
    make install && \
    ldconfig

# Build static wxWidgets:
# Note: replace '~/Develop/wxWidgets-staticlib' with your own path if you prefer, remember it to be used later when building CubicSDR.

RUN wget https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.3/wxWidgets-3.1.3.tar.bz2
RUN tar -xvjf wxWidgets-3.1.3.tar.bz2  && \
    cd wxWidgets-3.1.3/  && \
    mkdir -p ~/Develop/wxWidgets-staticlib && \
    ./autogen.sh && \
    ./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x" && \
    make -j4 && \
    make install

# Build CubicSDR
# Note: add -DUSE_HAMLIB=1 to cmake command line to include hamlib support.
RUN git clone https://github.com/cjcliffe/CubicSDR.git
RUN apt-get install -y libasound2-dev
RUN cd CubicSDR && \
    mkdir build && \
    cd build && \
    cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config -DUSE_AUDIO_ALSA=ON -DUSE_AUDIO_PULSE=OFF && \
    make -j4
```

Barbie2013, 2020-09-10:
Hi all, 

i ran into the same issue under Ubuntu 20.04.01.

 cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Dev/wxWidgets-staticlib/bin/wx-config
64 bit compiler detected
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:170 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Copying /home/pi/Dev/CubicSDR/font/*.fnt to directory /home/pi/Dev/CubicSDR/build/x64/fonts
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono16.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono27.fnt
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono36.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono64.fnt
-- Configuring file vera_sans_mono72.fnt
-- Configuring file vera_sans_mono96.fnt
-- Copying /home/pi/Dev/CubicSDR/font/*.png to directory /home/pi/Dev/CubicSDR/build/x64/fonts
-- Configuring file vera_sans_mono12_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono27_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono36_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono64_0.png
-- Configuring file vera_sans_mono72_0.png
-- Configuring file vera_sans_mono96_0.png
-- Copying /home/pi/Dev/CubicSDR/icon/CubicSDR.ico to directory /home/pi/Dev/CubicSDR/build/x64
-- Configuring file CubicSDR.ico
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIQUID_LIBRARIES
    linked by target "CubicSDR" in directory /home/pi/Dev/CubicSDR

-- Configuring incomplete, errors occurred!
See also "/home/pi/Dev/CubicSDR/build/CMakeFiles/CMakeOutput.log".

TJNII, 2021-06-09:
Retried this today and it passed.  Assuming it was fixed in one of the updates since I opened this.

closed 2021-06-09 by TJNII