cjcliffe/CubicSDR #850: Installing CubicSDR under Unbuntu, cannot find -lliquid

pjparis, 2020-09-10:
Hi--I'm trying to install CubicSDR 0.2.5 from source on a Raspberry Pi 4b (2 GB) running 
Ubuntu MATE 20.04, and I'm having a problem. Before going into detail about what 
I've done, I'll get straight to the point and hand off the [relevant] part of the error that's preventing 
me from completing the install:

...
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestCanvas.cpp.o
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
[100%] Linking CXX executable x86/CubicSDR
/usr/bin/ld: cannot find -lliquid
/usr/bin/ld: cannot find -lliquid
/usr/bin/ld: cannot find -lliquid
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1353: x86/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
$

If more details would help I'm happy to provide cmake's entire output. I just
didn't want to overload the message with more than might be necessary to 
interpret what's happening, or not.

The receiver is the SDRPlay RSP1a. 

Here's the command that is responsible for the lines above:

cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=/home/paul/Downloads/wxWidgets-3.1.3/Develop/wxWidgets-staticlib/bin/wx-config -DLIQUID_LIBRARIES=/home/paul/Downloads/liquid-dsp/include/liquid.h -DOTHER_LIBRARIES="-latomic" -DLIQUID_INCLUDES=/home/paul/Downloads/liquid-dsp/include/liquid.internal.h -DOTHER_LIBRARIES="-lliquid"

My install follows the method detailed at: https://cubicsdr.readthedocs.io/en/stable/building-cubicsdr/linux.html

SoapySDR, SoapySDRPlay, SDRPlay's drivers all seem to have installed w/o issue. I
can run SoapySDRUtil and find and interrogate the RSP1a, no problem.

wxWidgets seem to have installed problem-free, too. Ditto liquid-dsp...maybe.

My problem only begins with CubicSDR itself. I struggled with getting cmake to recognize
the liquid libraries during cmake, ending up with a modified cmake command string
that includes the -DLIQUID_LIBRARIES= and -DLIQUID_INCLUDES=. I tried the 
-DOTHER_LIBRARIES="-lliquid" in the hope that it might do the trick. Nope. 
 
Can anyone suggest a way around my problem(s)? I'm guessing that lliquid is a
valid reference to some liquid-dsp component, but could be wrong. Could my 
choice of OS (Unbuntu vs Raspbian) be at the root of my problem? Any assistance
is greatly appreciated. 

Thanks a bunch 

Paul

Dantali0n, 2020-09-10:
When installing from source on a Linux OS it is best to follow this guide: https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

This line will cause the compilation to fail `-DLIQUID_LIBRARIES=/home/paul/Downloads/liquid-dsp/include/liquid.h` as you are pointing to the header for liquid-dsp while you should provide the library folder here. The aforementioned guide shows how to properly install the required libraries so that cmake can automatically discover them.

pjparis, 2020-09-11:
DantaliOn---Thank you very much, both for the quick reply and the guidance. The instructions worked! I ran into only one minor issue with the install and that had to do with the line:

CFLAGS="-march=native  -03" ./configure --enable-fftoverride

this didn't work as written, but if I set the CFLAGS var first and then ran ./configure it was fine. After that, the CubicSDR make was successful

Thank you again. 

Paul

Dantali0n, 2020-09-11:
@vsonnier could you close this for me?

closed 2020-09-11 by pjparis