cjcliffe/CubicSDR #855: Screen Not displaying correctly on 0.2.6a (SDR Play version)

Paul-Nuneaton, 2020-10-07:
[image]
[image]
I have uploaded two screen shots of two different versions running on the same Mac. 0.2.4 looks perfect and the other, 0.2.6a is messed up.

To select anything I need to point and click where it should be, not where it actually is now.

These shots are from the internal Macbook screen. When I plug in my HDMI 4K monitor, the 0.2.6a version looks normal.

I am running Big Sur on a 2019 4 port Macbook Pro

FotisGR, 2020-10-09:
Exactly the same problem here on my macbook pro 2018's screen (all resolutions tested) with Catalina 10.15.7. 
Version 0.2.5 its without this problem

Dantali0n, 2020-10-09:
This is a duplicate of https://github.com/cjcliffe/CubicSDR/issues/727 please continue the discussion there. In addition, please test patch https://github.com/cjcliffe/CubicSDR/pull/769 if you have time as it solves this issue.

@vsonnier could you close this for me?

vsonnier, 2020-10-09:
Thanks @Dantali0n for your support ! Let's close this indeed.

closed 2020-10-09 by vsonnier