cjcliffe/CubicSDR #857: OSX 0.2.6a User Interface Controls Issue

josepharmbruster, 2020-10-30:
I just updated from 0.2.4 to 0.2.6a on macOS Catalina 10.15.7, because 0.2.4 did not appear to recognize my new SDR Duo.  After installing 0.2.6a, it recognizes my duo, however... the window controls are all appearing mis-sized.  See the attached:
[image]

Dantali0n, 2020-10-31:
This is a duplicate of #727 please continue the discussion there. In addition, please test patch #769 if you have time as it solves this issue.

vsonnier, 2021-08-02:
#769 has been merged, which should fix the problem.

closed 2021-08-02 by vsonnier