cjcliffe/CubicSDR #860: CubicSDR freezes with RSP2 after starting device

SietseAchterop, 2020-11-08:
Installation problem with CubicSDR and RSP2 device
====================================================

CubicSDR freezes after starting device.
Problem looks like issue #735.

Here is what I did, starting from installation guide from https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

- PC current Debian testing (november 8-th 2020)
- soapysdr   from debian  (libsoapysdr-dev version 0.7.2-1)
- liquid-dsp from debian  (libliquid-dev version 1.3.2-2)
- wxWidgets from source  (3.1.3)
- Cubicsdr     from github
- SoapyRTLSDR  from github
- SoapySDRPlay from github

Everything OK so far.

Output of SoapySDRUtil --info
=============================
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-1
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr

Search path:  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7
Search path:  /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.7               (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libmiriSupport.so   (0.2.5)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libremoteSupport.so (0.5.2)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so            (0.3.1-bec4f05)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so           (0.3.0-8c4e330)
vailable factories... miri, remote, rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

Note: modules both from debian testing and locally installed.
      could that be a problem?

Start Cubicsdr.
Select SDRplay device which is properly recognized: SDRplay Dev0 RSP2 1710046220

Everything looks ok until starting the device from Menu -> File -> Start/Stop Device.
After a few seconds Cubicsdr freezes completely with  the following message in the console:
  ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)

 What could be the problem?

   Thanks in advance,
       Sietse Achterop

=======================================
Console output after starting Cubicsdr
=======================================

Loading:: configuration file '/home/sietse/.CubicSDR/config.xml'

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.7.1
	ABI Version: v0.7
	Install root: /usr
	Loading modules... 
	Available factories...miri, null, remote, rtlsdr, sdrplay
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1710046220
  serial = 1710046220
Make device 0
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: Yes
	DRI2SwapInterval: No
	glxSwapIntervalMESA: Yes
	glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2
  serial=1710046220

Reporting enumeration complete.
SDR enumerator done.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono36_0.png', parsed 255 characters.
ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)
RtAudio pulse: _NOT_ running realtime scheduling
SDR thread starting.
device init()
[INFO] Using format CF32.
SDRThread : Device Stream set to MTU: 65536
starting readLoop()

ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)

vsonnier, 2020-12-25:
I've experienced a similar case on Windows x64 where the UI freezes but the audio continues to run. Once I upgraded to Wxwidgets 3.14, the problem desappeared. 

However : 

> Everything looks ok until starting the device from Menu -> File -> Start/Stop Device.
After a few seconds Cubicsdr freezes completely with the following message in the console:
ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)

What could be the problem?

Your device is already active after you select it in the initial Device window. Going to the Menu again actually stops it.

vsonnier, 2021-03-27:
We have countless examples of RSPs working great on Linux, for instance here: https://github.com/pothosware/SoapySDRPlay3/issues/23 but both CubicSDR and SoapySDRPlay must be built from `master`. 

In particular, you now need SDR API v3+ for the latest and greatest `SoapySDRPlay` module.

And Cubic must be built with the latest wxWidgets, i.e v3.14 at the time of writing.

closed 2021-03-27 by vsonnier