cjcliffe/CubicSDR #861: MacOS builds not working anymore with wxWidgets 3.1.4

vsonnier, 2020-11-11:
Hi @cjcliffe,

I woke up for a loooonnng sleep to update the builds to wxWidgets 3.1.4 released in July, unfortunatly the current compiling options apparently no longer works for MacOS build, as the Travis build shows. 
Everything is fine for Windows and Linux on the other side. 

Since I don't have MacOS, I've no idea what to do next.

SDRplay, 2020-11-11:
Can you confirm if the issue is building wxWidgets 3.1.4 itself or building CubicSDR with wxWidgets 3.1.4 already built?

I have a Mac I can test with.

Andy

vsonnier, 2020-11-11:
Hi Andy, 

According to the Travis build log: https://travis-ci.org/github/cjcliffe/CubicSDR/jobs/742902112 (at the end) 
it is failing in the wxWidgets building phase.

SDRplay, 2020-11-11:
Hi Vincent,

The error is complaining about the -mmacosx-version-min=10.9 option and the alternatives are to either use a different minimum OS version or omit it completely. I've omitted it completely for now and for me it has defaulted to 10.10 and it's all building ok.

10.9 is Maverick
10.10 is Yosemite

Personally I don't think that's a problem, but it looks like Maverick will no longer be supported by wxWidgets from 3.1.4 onwards unless that's a bug for the wxWidgets team to look at?

Andy

vsonnier, 2020-11-11:
> I've omitted it completely for now and for me it has defaulted to 10.10 and it's all building ok.

Let's assume we can remove that option then, and see if users still use 10.9 out there and have problems. Thank you for your test !

vsonnier, 2020-11-11:
Andy, I've commited the suggested change, and the Travis build completed successfully. I'll keep this issue open though, to track further problems.

SDRplay, 2020-11-11:
Sounds good 👍

ah1102, 2020-11-30:
Add 0.2.6 release for mac, linux. At least alpha. How long has passed and there is no release. What a pity that such a promising project is practically not developing. Perhaps the plugin system could somehow develop the project in the absence of the author-developer, but, unfortunately, he did not lay this principle. Take sdr #, this program is worse than cubicSDR, but sdr # has a plugin system and thanks to it the project is developing rapidly.

ah1102, 2020-11-30:
I am really looking forward to the release of the SDRuno v2 project with mac and linux support, despite the fact that the project is proprietary. SDRuno can seriously overshadow sdr # if it makes its project multiplatform.

vsonnier, 2020-12-25:
Time to close this.

closed 2020-12-25 by vsonnier