cjcliffe/CubicSDR #863: Is it possible to build CubicSDR without GLX dependencies?

ka2ddo, 2020-12-06:
I'm trying to access a remote SDR on a Raspberry Pi over VNC. However, trying to start the Raspbian distro's version of CubicSDR fails in the VNC server/client environment, complaining that it can't get the GLX version number over and over again; specifically, 

./src/unix/glx11.cpp(589): assert "ok" failed: in GetGLXVersion(): GLX version not found

It works fine from a monitor plugged into the HDMI port on the Pi itself.

So, is it possible to build CubicSDR to not depend on GLX, so I can access it remotely?

Dantali0n, 2020-12-07:
You could try removing `--with-opengl` from the wxWidgets configuration.

```
./configure --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x"
```

To be able to test this you will have to follow the complete build guide on the wiki, you can't use the AppImage:
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

vsonnier, 2020-12-07:
CubicSDR needs OpenGL in wxWidgets, simply because it is an OpenGL application. While it is definitly possible @ka2ddo, the correct way to do it in your case is to install SoapyRemote on the PI and leave the CubicSDR on your local computer.

vsonnier, 2020-12-25:
Closing this, by lack of convsersation. There is actually 2 way of runing SDR with Pi:
-  Runing CubicSDR on the Pi, provided you have configured OpenGL hardaware acceleration (ex : https://www.youtube.com/watch?v=dx_iEpcZsDs )
-  Runing SoapyServer on the Pi to use it remotely on a separate CubicSDR instance through the SoapyRemote module. (see https://github.com/pothosware/SoapyRemote/wiki )

Both solutions have already been explained at length on existing CubicSDR issues, or elsewhere in the Internet.

closed 2020-12-25 by vsonnier