cjcliffe/CubicSDR #864: Pls release 0.2.6a for macOS

ah1102, 2020-12-06:
Please release 0.2.6a for macOS. There has been no release for a very long time, although the program has undergone many improvements.

cjcliffe, 2021-02-23:
Will do; I've been on crunch time since last August for a few projects and really haven't had time to do work on any open source stuff or release QA.

Just going to fix automatic build and some cross-platform screen scaling issues and then I'll consider it ready for release.

Hi-Tone, 2021-03-18:
We are a lot who is dying to get a release which also supports Retina display MacBooks, so we can enjoy SDR on the run. Thanks.

dmaltsiniotis, 2021-03-18:
@Hi-Tone I have a mid-2015 Macbook pro with Retina display and CubicSDR runs just fine. Is there something else that prevents CubicSDR running for you, or are you referring to not using UI scaling and instead native support for the resolution?

Hi-Tone, 2021-03-19:
@dmaltsiniotis I think that might be the case. I do see a lot of options missing, but that might also have something to do with the UI scaling. I have a 2017 MacBook Pro. How do I change the UI scaling, so I can see everything normally ?

snwvo7, 2023-01-15:
I have the same UI scaling issue on a MacBook Air M1 running Ventura 13.1
Is this already fixed?

vsonnier, 2023-01-16:
It looks the same issue as #727, mark this as duplicate. Please note that only `0.27` version built against the latest `wxWidgets` (3.21) will be considered for support, since Cubic is a hobby project.

closed 2023-01-16 by vsonnier