cjcliffe/CubicSDR #868: Empty modem settings

PopeFelix, 2021-01-10:
I'm running CubicSDR with an RTL-SDR RTL2832 USB dongle. I can see the waterfall, but I can't hear anything. Looking through previous issues suggests that I should be able to do that under "Modem settings", but my "Modem settings" is empty and clicking on it produces no reaction. If I click on the waterfall, I'm able to choose a frequency, but it doesn't change my modem settings.

alexchukanov, 2021-02-04:
the same problem!
What is more, they write about mir_sdr_api.dll file in sdrplay dir, but it is absent there!

SDRplay, 2021-02-04:
The mir_sdr_api.dll you would get if you installed the SDRplay API 2.13, but the OP is talking about an RTL-SDR device so probably not relevant to @PopeFelix's issue.

colbyAtCRI, 2021-02-05:
Sorry for the basic question. Have you first selected a modem AM, LSB, FMS etc and a demod frequency by clicking (placing a demod region) on the waterfall? Modem settings (which are sometimes hidden collapsed to the right of the modem selection bar) are empty until these are selected.

alexchukanov, 2021-02-05:
Thank you Paul, I have already solved my problem!
Alex.

PopeFelix, 2021-02-05:
I don't recall, but I'll confirm.

PopeFelix, 2021-02-07:
Here's what happens:

I start CubicSDR
I select a device (in this case, a generic RTL-SDR)
I choose AM
I set Frequency to 15 MHz, Bandwidth to 6 kHz, and Center Frequency to 15
MHz.
"Modem Settings" is empty.

vsonnier, 2021-02-08:
@PopeFelix You know, you can attach screenshots to GitHub issues, provided you use the Web interface. Also, please reply with the Web interface, replying to mails are very limited.

> "Modem settings" is empty

I bet your RTL-SDR is not detected at all. In any case, screenshots.

vsonnier, 2021-03-02:
Closed by lack of feeback.

closed 2021-03-02 by vsonnier