cjcliffe/CubicSDR #870: wxWidgets not visible OrangePi Plus 2e

nsklyarov, 2021-01-15:
Hi, i try compile CubicSdr on 
OrangePi Plus 2e 
Ubuntu 16.04.7 xenial
armv7l
according to wiki instructions.

all steps go through without errors, but when I launched Cubic and selected the required device, Cubic did not display Waterfall and other widgets:

[image]

Please tell me what could be the problem? Thanks

nsklyarov, 2021-01-15:
maybe the case is in libGL.so, cannot find this library on the system

vsonnier, 2021-01-15:
> maybe the case is in libGL.so, cannot find this library on the system

Of course it is, CubicSDR is an OpenGL application : you need WxWidgets compiled with OpenGL support, AND your Pi must have Hardware acceleration.

nsklyarov, 2021-01-27:
Yes you are right. the problem is not in the assembly, the problem is that I ran the application through VNC. VNC does not support openGL, but the board itself supports. if I use the board directly - then all is well. Thank you for your prompt reply.

closed 2021-01-27 by nsklyarov