cjcliffe/CubicSDR #872: What is the data format of I/Q audio output?

tkzic, 2021-01-29:
Looking for documentation about the data format of I/Q output from CubicSDR. eg., CS8, CS16, etc.,

Thanks,
Tom
KA1IS

vsonnier, 2021-01-29:
There is no IQ output in Clubic.

dmaltsiniotis, 2021-01-29:
According to the manual here https://cubicsdr.readthedocs.io/en/latest/application-window.html (and my own usage experience) I think @tkzic may be referring to the I/Q "de-modulator", which you can then record by pressing R to a wav file.

> I/Q: Raw I/Q Pass-Thru (No Modulation)
> Raw I/Q samples that would normally go to a modem are passed through to the sound card for use elsewhere. Bandwidth is fixed to the selected sound card output frequency and will change along with it. Note that turning the Audio Gain down to a low level will disable gain completely and *output the raw decimated samples.*

I think @tomswartz07 may be asking is: What is the IQ file format of this "I/Q de-modulator" output when recording it?

I actually have had the same question for some time.
[image]

vsonnier, 2021-01-29:
A right, this one.  

Apparently, it generates 2 F32  audio sample per baseband-CF32, simulating Stereo: le Left channel being = Q (im) and Right Channel = I (real) part. 
The resulting audio sample rate = the demod bandwith, unless I'm mistaken.

tkzic, 2021-01-29:
Thanks vsonnier. That's exactly what I was looking for.
Tom
KA1IS

closed 2021-01-29 by tkzic