cjcliffe/CubicSDR #874: Failed on Pi 4

DasClarky, 2021-02-08:
pi@raspberrypi:~/CubicSDR/build $ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  3%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
/home/pi/CubicSDR/src/AppFrame.cpp: In member function ‘bool AppFrame::actionOnMenuSettings(wxCommandEvent&)’:
/home/pi/CubicSDR/src/AppFrame.cpp:1502:50: warning: catching polymorphic type ‘class std::invalid_argument’ by value [-Wcatch-value=]
                     catch (std::invalid_argument e) {
                                                  ^
/home/pi/CubicSDR/src/AppFrame.cpp:1519:50: warning: catching polymorphic type ‘class std::invalid_argument’ by value [-Wcatch-value=]
                     catch (std::invalid_argument e) {
                                                  ^
/home/pi/CubicSDR/src/AppFrame.cpp: In member function ‘bool AppFrame::actionOnMenuRig(wxCommandEvent&)’:
/home/pi/CubicSDR/src/AppFrame.cpp:1845:48: warning: unused parameter ‘event’ [-Wunused-parameter]
 bool AppFrame::actionOnMenuRig(wxCommandEvent &event) {
                                ~~~~~~~~~~~~~~~~^~~~~
[  4%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o
/home/pi/CubicSDR/src/AppConfig.cpp: In member function ‘void DeviceConfig::load(DataNode*)’:
/home/pi/CubicSDR/src/AppConfig.cpp:223:31: warning: ‘fltSettingValue’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             if (keyName != "" && !(fltSettingValue!=fltSettingValue)) {
                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  5%] Building CXX object CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o
/home/pi/CubicSDR/src/FrequencyDialog.cpp: In constructor ‘FrequencyDialog::FrequencyDialog(wxWindow*, wxWindowID, const wxString&, DemodulatorInstancePtr, const wxPoint&, const wxSize&, long int, FrequencyDialog::FrequencyDialogTarget, wxString)’:
/home/pi/CubicSDR/src/FrequencyDialog.cpp:16:138: warning: unused parameter ‘position’ [-Wunused-parameter]
 FrequencyDialog::FrequencyDialog(wxWindow * parent, wxWindowID id, const wxString & title, DemodulatorInstancePtr demod, const wxPoint & position,
                                                                                                                          ~~~~~~~~~~~~~~~~^~~~~~~~
/home/pi/CubicSDR/src/FrequencyDialog.cpp:17:24: warning: unused parameter ‘size’ [-Wunused-parameter]
         const wxSize & size, long style, FrequencyDialogTarget targetMode, wxString initString) :
         ~~~~~~~~~~~~~~~^~~~
/home/pi/CubicSDR/src/FrequencyDialog.cpp: In member function ‘void FrequencyDialog::OnChar(wxKeyEvent&)’:
/home/pi/CubicSDR/src/FrequencyDialog.cpp:173:32: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
             } catch (exception e) {
                                ^
/home/pi/CubicSDR/src/FrequencyDialog.cpp:188:32: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
             } catch (exception e) {
                                ^
/home/pi/CubicSDR/src/FrequencyDialog.cpp:204:32: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
             } catch (exception e) {
                                ^
In file included from /usr/include/c++/8/map:60,
                 from /home/pi/CubicSDR/src/demod/DemodulatorInstance.h:7,
                 from /home/pi/CubicSDR/src/FrequencyDialog.h:10,
                 from /home/pi/CubicSDR/src/FrequencyDialog.cpp:4:
/usr/include/c++/8/bits/stl_tree.h: In member function ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_emplace_hint_unique(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, _Args&& ...) [with _Args = {const std::piecewise_construct_t&, std::tuple<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>, std::tuple<>}; _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >]’:
/usr/include/c++/8/bits/stl_tree.h:2411:7: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >’} changed in GCC 7.1
       _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/map:61,
                 from /home/pi/CubicSDR/src/demod/DemodulatorInstance.h:7,
                 from /home/pi/CubicSDR/src/FrequencyDialog.h:10,
                 from /home/pi/CubicSDR/src/FrequencyDialog.cpp:4:
/usr/include/c++/8/bits/stl_map.h: In member function ‘void FrequencyDialog::OnChar(wxKeyEvent&)’:
/usr/include/c++/8/bits/stl_map.h:499:8: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >’} changed in GCC 7.1
    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
/usr/include/c++/8/bits/stl_map.h:499:8: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >’} changed in GCC 7.1
    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
/usr/include/c++/8/bits/stl_map.h:499:8: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >’} changed in GCC 7.1
    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
/usr/include/c++/8/bits/stl_map.h:499:8: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, SoapySDR::Range> >’} changed in GCC 7.1
    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
[  6%] Building CXX object CMakeFiles/CubicSDR.dir/src/DemodLabelDialog.cpp.o
[  8%] Building CXX object CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o
[  9%] Building CXX object CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o
/home/pi/CubicSDR/src/ModemProperties.cpp: In member function ‘wxPGProperty* ModemProperties::addArgInfoProperty(wxPropertyGrid*, ModemArgInfo)’:
[... cut ...]
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x48f0): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x4a8c): undefined reference to `__atomic_load_8'
/usr/bin/ld: SoapySDRThread.cpp:(.text+0x4aa0): undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1353: x86/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

A bit stumped here to be honest!

vsonnier, 2021-02-08:
Duplicate of #809 .

closed 2021-02-08 by vsonnier

colbyAtCRI, 2021-02-08:
Looks like you might be missing #include <atomic>  or some associated lib?

DasClarky, 2021-02-08:
-DOTHER_LIBRARIES="-latomic"

i need to add that to the cmake line, i didn't notice the other thread so apologies for dup posting and thanks @vsonnier