cjcliffe/CubicSDR #875: Airspy HF+ Discovery not seen when running CubicSDR 0.2.4

Bromo33333, 2021-02-14:
I downloaded the Darwin package of 0.2.4 and it is running on my Mac (Macbook Pro, MacOS 11.2.1 Big Sur), but it does not see my Airspy HF+ Discovery.  I see the Airspy on the USB Bus, but for some reason it does not register.

I am sure it is something simple, that I am just missing, but I couldn't find any documents saying what to look at or do in this case.  Any jelp would be appreciated!

ethan-sturm, 2021-04-29:
I am having the exact same issue with a new Airspy HF+ Discovery on my Mac mini. Were you ever able to get this resolved?

vsonnier, 2022-01-15:
Version `0.24` is waayyy too old. It is an issue on a specific device anyway so the issue is on https://github.com/pothosware/SoapyAirspyHF

closed 2022-01-15 by vsonnier