cjcliffe/CubicSDR #876: Windows build support for Travis-CI

cjcliffe, 2021-02-23:
Windows support appears to have progressed on Travis-CI; add support for it

cjcliffe, 2021-04-30:
new travis-ci model makes it impossible to use for CubicSDR; single build eats all the credits.  Removing Travis-CI support and will look for alternatives

closed 2021-04-30 by cjcliffe

Dantali0n, 2021-05-02:
Personally I have installed Gitlab and Gitlab-runner locally in my network on a machine. This can be accomplished using docker reducing the potential impact on your home network.

A simple domain name is enough to allow anyone outside your home network access with granular control. The Gitlab docker container can automatically use letsencrypt to renew its certificate. 

Gitlab-runner can be configured to enable multi environments, there is a pretty well defined setup for virtualbox + Windows. Mac should be possible as well but is more difficult than the Windows solution.

The whole setting up infrastructure in your home network and exposing it to the internet is non-ideal but I believe this is currently the only method to have a single free solution for Linux + Windows + Mac builds.