cjcliffe/CubicSDR #878: Specifying "Other Settings" for SoapySDR

wmszeliga, 2021-03-01:
I have an SDRplay RSP1A with the Soapy modules v0.8 from SDRplay. Probing the device with SoapySDRUtil yields the following output
```######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] devIdx: 0
[INFO] hwVer: 255

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=20060DD397
  sdrplay_api_api_version=0.000000
  sdrplay_api_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.125, 0.25, 0.5, 1, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```
As you can see, there are "Other Settings" available such as RF Gain Select and BiasT enable/disable. Is it possible to get CubicSDR to set/enable/disable these "Other Settings"? I don't see them on the setting menu and am unsure of how I would specify them in `~/Library/Application Support/CubicSDR/config.xml`, if that's even a possibility. I'm running CubicSDR 0.2.6a (from SDRplay) along with the SDRplay provided v0.8 Soapy modules. All of this is on a Mac running 10.15.7.

vsonnier, 2021-03-01:
> As you can see, there are "Other Settings" available such as RF Gain Select and BiasT enable/disable. Is it possible to get CubicSDR to set/enable/disable these "Other Settings"? I don't see them on the setting menu

They should be definitly there. Screenshots ?

wmszeliga, 2021-03-02:
OK, I've attached a screenshot of the "Settings" menu as it looks with the RSP1A device started.

vsonnier, 2021-03-02:
Thanks, we also need to see the `SDR Drvices` initial dialog, to really see what device is actually selected, with its initial settings.

wmszeliga, 2021-03-02:
[image]

vsonnier, 2021-03-02:
Your device is definitely not detected. You should open a support ticket to @SDRPlay, there is nothing much that can be done on the CubicSDR side.

closed 2021-03-02 by vsonnier