cjcliffe/CubicSDR #883: No audio but waterfall is present. Cannot create modem.

scottbomb, 2021-04-03:
Fresh install on Kubuntu 20.04. This has worked before on the same hardware, same OS. When I launch the program, I see and select my dongle. I see a waterfall showing an FM radio station at 99.7. I click the waterfall and nothing happens. I click FM on the top-left and it turns gray but still no sound. There isn't anything under Modem Settings, the box is blank. 

Here's a screenshot: http://n0xmz.com/image/cubicsdr.png 

Any ideas on what I'm missing? I open pulseaudio volume control and CubicSDR doesn't show up anywhere. I'd expect to see it on the Playback tab where other audio-playing applications (like VLC) usually appear.

vsonnier, 2021-04-23:
First, your version `0.25` is waaaayy too old, please build from `master`. Second, you indeed created a modem as the screeshot shows but you have no sound likely because that version was not compiled with support of PulseAudio in RtAudio. 
Please follow the Wiki instructions to compile a version from sratch eventually modifying the audio support options as suggested here #702

vsonnier, 2021-05-07:
Closing by lack of feedback.

closed 2021-05-07 by vsonnier