cjcliffe/CubicSDR #886: no window refresh/update with wxWidgets-3.1.5 (Gtk+-3 backend)

stefantalpalaru, 2021-04-16:
[image]

vsonnier, 2021-04-16:
Either your wxWidgets was compiled without OpenGL support, or you don't have proper OpenGL. Both are mandatory. Have a look at the Wiki.

stefantalpalaru, 2021-04-16:
> Either your wxWidgets was compiled without OpenGL support, or you don't have proper OpenGL.

```text
$ lddtree /usr/bin/CubicSDR
...
    libwx_gtk3u_gl-3.1-gtk3.so.5 => /usr/lib64/libwx_gtk3u_gl-3.1-gtk3.so.5
...
        libEGL.so.1 => /usr/lib64/libEGL.so.1
            libGLdispatch.so.0 => /usr/lib64/libGLdispatch.so.0
...
    libGL.so.1 => /usr/lib64/libGL.so.1
        libGLX.so.0 => /usr/lib64/libGLX.so.0
```

stefantalpalaru, 2021-04-17:
Configuring wxWidgets-3.1.5 with `--disable-glcanvasegl` fixed the problem.

Dantali0n, 2021-04-17:
> Configuring wxWidgets-3.1.5 with `--disable-glcanvasegl` fixed the problem.

Do you think this should be added to the wiki or is this some edge case due to your Graphics cards its egl implementation being funky.

Perhaps you could indicate the hardware and accompanying driver being used.

stefantalpalaru, 2021-04-17:
> Do you think this should be added to the wiki or is this some edge case due to your Graphics cards its egl implementation being funky.

I think it's a matter of the CubicSDR GLX context competing with a wxWidget's EGL context. I'm no expert, but from a summary reading, it looks like you can't mix the two in the same application.

Sure, you can document that in the wiki for people building their own wxWidgets, but when 3.1.5 hits distro package repositories, with EGL built in by default, the problem is back.

A long-term solution would be to allow choosing between GLX and EGL in CubicSDR, at configuration time.

vsonnier, 2021-04-17:
> I think it's a matter of the CubicSDR GLX context competing with a wxWidgets EGL context. I'm no expert, but from a summary reading it looks like you can't mix the two in the same application.

Me neither, I just heard of EGL here... Since CubicSDR access OpenGL "raw" in the GL wxGL context, it then make sense of disable EGL in wxWigets as @stefantalpalaru found. I've changed the Wiki accordingly.

Thanks @stefantalpalaru for your contribution ! Thanks @Dantali0n for your support as well.

vsonnier, 2021-04-17:
I've updated the Travis scripts as well, referencing 3.1.5 with @stefantalpalaru fix.

vsonnier, 2021-04-23:
I think the issue can be closed now.

closed 2021-04-23 by vsonnier