cjcliffe/CubicSDR #889: wxWidgets 3.1.5 Build, Wonky Display in Cubic

ehidle, 2021-05-02:
I occasionally build the latest Cubic from github, but this time there's a catch and I am not sure if it's because of wxWidgets or something else. 

The cubic build instruction page at https://cubicsdr.readthedocs.io/en/stable/building-cubicsdr/linux.html instructs to configure wxWidgets with the following options:

./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid **--disable-webkit** --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x" --with-libiconv=/usr

However, **--disable-webkit** is not valid in 3.1.5

ehidle@w2jkt:~/software/wxWidgets$ ./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x" --with-libiconv=/usr
**configure: error: unrecognized options: --disable-webkit**

wsWidgets will build fine if I remove that option, and then Cubic will also build, but when I start the resulting binary, I get this: 

[image]

If I open one of the bookmarks, I get the audio I expect, but none of the visual features seem to work. 

I'm building wx from the 3.1.5 bzip and I tried building Cubic from both the current origin/master (bd7574a3) and the current origin/compiler-warning-cleanup-fixes (d0b4bf69) and got the same results. 

Thoughts/recommendations would be most welcome.
Thanks!

vsonnier, 2021-05-02:
Likely duplicate of #886, try the new method described there.

ehidle, 2021-05-02:
Thank you. Using the updated instructions in the Wiki, I still get the **configure: error: unrecognized options: --disable-webkit** for wxWidgets, but like before everything seems to build okay without that option. 

Other than that, this seems to have resolved the issue and I've updated my local build scripts to otherwise be in-line with the wiki.

Thanks again for all that you do!

closed 2021-05-02 by ehidle

vsonnier, 2021-05-07:
@ehidle  Sorry to ping, but what system are you running ? It may be intresting to update the Wiki with you findings about the `unrecognized options: --disable-webkit`.