cjcliffe/CubicSDR #898: CubicSDR on Ubuntu 21.04 Hangs GPU

4nanook, 2021-06-01:
In the last few weeks CubicSDR started hanging the GPU immediately upon start-up.  I see in the Launchpad site that I'm not the only one experiencing this.  It happens whether I use a self-compiled copy of the Ubuntu provided package.

Dantali0n, 2021-06-02:
Since similar reports are being made on launchpad that you have found, got you include these in this report?
Similarly we need more information to be able to resolve this issue. In principle always try to include information about the system and logs for any issue.

For logs this can be the output of CubicSDR but also the logs of Xorg or Wayland. Since it involves a GPU hang the distinction between if Xorg or Wayland is used can also be of value.

In addition some information about the OpenGL runtime and hardware. The easiest method to retrieve this is by installing `inxi` and sharing the output of `inxi -G` here.

Dantali0n, 2021-06-03:
Perhaps related is: #899

4nanook, 2021-06-03:
The GPU hang was a misconfiguration of my machine.  Still the CubicSDR from Ubuntu 21.04 core dumps but from 20.04 works.  I am going to try compiling myself on 21.04 to determine if this is just a Ubuntu issue or an issue with the program.

4nanook, 2021-06-04:
Ok, I've got it built from source and it's working with one minor problem, the mouse wheel is making the frequency digit go up or down by two instead of one which makes tuning problematic.

vsonnier, 2023-01-13:
Issue is too old, with too old CubicSDR, closing.

closed 2023-01-13 by vsonnier