cjcliffe/CubicSDR #911: AppImage Gives errors

sprakashb, 2021-08-03:
On Ubuntu 20.4 running AppImage produces..
..
(CubicSDR-0.2.5-x86_64.AppImage:4133): Gtk-WARNING **: 09:40:20.648: Theme parsing error: gtk.css:6385:15: Missing name of pseudo-class
Loaded 230 rig models via hamlib.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.7.0
	ABI Version: v0.7
	Install root: ./////////
	Loading modules... 

(CubicSDR-0.2.5-x86_64.AppImage:4133): GLib-CRITICAL **: 09:40:23.591: g_error_free: assertion 'error != NULL' failed
[ERROR] SoapySDR::loadModule(.//////////lib/SoapySDR/modules0.7/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
	Available factories...airspy, airspyhf, audio, bladerf, hackrf, lime, null, redpitaya, remote, rtlsdr
Found Rafael Micro R820T tuner
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = PulseAudio
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 1

(CubicSDR-0.2.5-x86_64.AppImage:4133): GLib-CRITICAL **: 09:40:24.046: g_error_free: assertion 'error != NULL' failed
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers

What is missing? The RTL- SDR works in GQRX.
Thanks for your wonderful work.

VU2SPF

FastThenLeft, 2021-08-08:
I get the same error:
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers

FastThenLeft, 2021-08-08:
I checked, and I have the current version:
libglx-mesa0 is already the newest version (21.0.3-0ubuntu0.2)

waterwin, 2021-08-12:
Maybe also check your OpenGL version. I had kind of the same issue on an older laptop HP 2540p with SDR++ as that needs OpenGL 3.0 or higher and now I have the same with the CubicSDr app image.
glxinfo | grep 'version'
Not sure what the OpenGL prerequisite for CubicSDR is.

FastThenLeft, 2021-08-12:
I installed the absolute latest, and that didn't fix it. FYI, I'm on Ubuntu 21.04.

cascmptrski, 2021-10-04:
Seeing exactly same behavior, Ubuntu 20.04 LTS with latest updates:

./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers

moof-moof, 2022-09-26:
Update:
Further investigation (mainly this gh reference [https://github.com/cjcliffe/CubicSDR/issues/697#issuecomment-452146000](url)   ) pointed to a pragmatic workaround that is good enough for me. I can now start the app via a shell script containing the commands:
`cd /home/USERNAME/path-to-app LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 ./CubicSDR-0.2.5-x86_64.AppImage`

N0rbert, 2022-10-02:
> Update: Further investigation (mainly this gh reference [https://github.com/cjcliffe/CubicSDR/issues/697#issuecomment-452146000](url) ) pointed to a pragmatic workaround that is good enough for me. [...]

For desktop shortcut above turns into *Command*:

```
env LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 /home/username/Software/AppImages/CubicSDR-0.2.5-x86_64.AppImage
```

Where `/home/username/Software/AppImages/CubicSDR-0.2.5-x86_64.AppImage` is a full path to AppImage.

cjcliffe, 2026-11-15:
New 0.3.0 builds will use a new base; closing this for now

closed 2026-11-15 by cjcliffe