cjcliffe/CubicSDR #913: Missing Modem Settings (No Sound)

mrspirytus, 2021-08-03:
For system and version information please check issue #912

I was able to get sound (sporadic) on the Generic RTL2832U OEM device. But never with HacKRF. I will explain in a few what "sporadic" I mean.

So I have both SDR devices connected. Then I start CubeSDR 0.2.5
For each device separate run, I select for example Generic RTL2832U OEM.
Then I adjust to the FM radio frequency range. 
At this point my "Modems Settings" list is empty and no Demodulator option is selected (to the least of it). Next, I select "FM" in the demodulator list and next I click on the bottom pane where the popup message states "Click to set demodulator frequency"

At this point in some cases, my "Modem Settings" will get set with Audio Device. This is what I mean by "sporadic". I quit CibeSDR and restart and repeated steps. At some point, I was able to get sound. Not sure if I am missing some step or I am doing something wrong. So please let me know.

No the other end, If I try to select HackRF device I was never able to get Audio Device.
I did make sure to grep in the terminal if there is another version of CubeSDR that is not running for some reason in the background. 

Please let me know what other information I can provide.
For now, I am running GQRX until I get this issue resolved. I prefer the CubeSDR interface :)
And yes, I do make sure GQRX is not running when starting CubeSDR.

Thanx

FinestDice17, 2021-12-01:
did anything come of this? Im having the same issue but I can not get my sdr to show up in cubicsdr at all after that first screen pops up and I select it there.

mrspirytus, 2021-12-02:
Nop. Still issue here :(

vsonnier, 2023-01-13:
Issue is too old, with too old CubicSDR, closing.

closed 2023-01-13 by vsonnier