cjcliffe/CubicSDR #916: Build doesn't work on Apple Silicon (M1) Mac

ecwhitney42, 2021-09-19:
The build for SoapySDR fails with architecture (referring to arm64).

My guess is something in the makefiles is trying to figure out the target architecture and it has never been tested with this one...

-Erich

SDRplay, 2021-12-24:
Building the SoapySDR framework for M1 is straight-forward and that all works well. However, my CubicSDR M1 build refuses to start - anyone got it working with wxWidgets 3.1.5?

Andy

SDRplay, 2021-12-24:
Figured it out - Hamlib was causing the issue - don't know why - maybe it's a x64 build that homebrew installs?? I'll try building it from source to see if that fixes the problem. Other than that, CubicSDR seems to be running fine on M1 so I think this issue can be closed.

vsonnier, 2022-01-15:
Thanks @SDRplay for your support and findings ! Closing !

closed 2022-01-15 by vsonnier